Ultimaker/Cura #7323: Preview not functional.

marave, 2020-03-21:
**Application version**
4.4, 4.5

**Platform**
Windows 10, RTX2060

**Printer**
Custom

**Reproduction steps**
1. Went to Preview tab

**Screenshot(s)**
[image]

**Actual results**
Preview doesn't show layers and moving the slider does nothing.

**Expected results**
Should show layers.

**Project file**
Happens on all projects.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4364328/cura.log)

**Additional information**
Works fine on my MacBook.

Ghostkeeper, 2020-03-24:
I'm unable to reproduce this problem:
[image]

You could try to go in the preferences and check the "Force layer view compatibility mode" option? Does that help any?

The log reveals that your GPU claims it supports OpenGL4, which is all it should need.

marave, 2020-04-17:
That does improve things.  Doesn't work quite right though.  Thanks

[image]

closed 2020-04-17 by marave

Ghostkeeper, 2020-04-22:
Yeah, the compatibility mode has fewer features and looks a bit dated. Sorry.