Ultimaker/Cura #7324: [FR] Smoother first layer when it contains holes

yet-another-average-joe, 2020-03-22:
**Is your feature request related to a problem? Please describe.**

When the first layer contains small holes (for example for small screws), the first layer is not eye candy because of the hole walls : concentric patterns ; holes cannot be printed without traveling in skin.

**Describe the solution you'd like**

Cura could offer to "close" the 1st layer holes. Later, the user will just have to use a drill or a blade, and open them, to get a gorgeous look

**Describe alternatives you've considered**

I close the holes in the CAD software with a layer equal to the Cura 1st layer height. It is not a problem for the model creator, but this trick is not really usable when releasing a model for other people : the first layer depends on material, nozzle, user habbits, etc. . For example, with PETG and 0.5mm nozzle, I use 1st layer = 0.3mm, but someone else could want to use PLA with a 0.4mm nozzle and a 0.12mm 1st layer...

Such a feature would need some sort of filter, closing holes depending on their area.

Another alternative could be a "special raft" : 1 layer, as thin as possible (using the 1st layer parameters in the Quality pane. I failed attempting to play with raft parameters... Maybe the parameters I used were wrong.

But in this case, the dimensionnal accuracy is a problem.

Another idea : no walls in the first layer ; no idea what problems this could cause... This is very different, but maybe it could be inteeresting to experiment with.

**Affected users and/or printers**

All printers

**Additional context**

1st layer with small holes (4.5mm)

[image]

1st layer with  closed holes :

[image]

 2nd layer with closed holes :

[image]

smartavionics, 2020-03-22:
My Cura builds have a travel option that allows you to ask for z-hop only on the first layer so it can stop those ugly streaks on the bottom skin.

yet-another-average-joe, 2020-03-22:
@smartavionics : this is not really about the streaks, it is about the concentric patterns around the holes. No moves in this case is just a side effect. Also, having a uniform linear pattern makes it nearly invisible when printed on glass with a new lack coating : ideal for front panels for example.

Also, small holes on the 1st layer are not really a problem with PLA, but they are when printing PETG  : they are often messed up or even dragged by the nozzle, ruining the whole 1st layer with blobs here and there (and it was the reason why I started to close the holes).

I know your branch : it's been installed on my PC, along with the official release, since 20190607 release. I have to say I hesitated posting here or on your branch...

(once again, it's not an on purpose request for personal use, as I design my own models : closing holes is done easily and quickly at CAD level : 60 seconds at most for 4 holes, STL export included)

smartavionics, 2020-03-22:
I often make holes blank and drill them out later. I most often do that for horizontal holes rather than vertical holes which you are concerned with. Here's an example, it has 4 horizontal holes all of which have a wall across them so that the outer wall of the model is printed continuously which improves the quality.

[image]

yet-another-average-joe, 2020-03-22:
Oh yesssss !!! Thanks for the trick !

Ghostkeeper, 2020-03-24:
You can also achieve this right now by placing a cutting mesh over the first layer that enables "Remove all holes":
[image]

In many ways this request is similar to "sacrificial bridging".

yet-another-average-joe, 2020-03-25:
I just tested on a model. Unfortunately, it kills the nice 1st layer pattern even setting walls width to zero, and it seems to me that the cutting mesh Z position/height cannot be set automatically to the 1st layer height ; or I'm missing something... (and it's really difficult to position accurately)

[image]

The problem is the 1st layer height. Imagine a model designed for PLA, 0.4mm nozzle, 1st layer 0.1. The holes are closed at design level, with a 0.1mm sacrificial skin. Someone else wants to print it with PETG, and a 0.5mm nozzle, 0.3mm 1st layer (love this combination). The slicer will ignore the "sacrificial skin". (I always failed setting 1st layer at less than 0.3 with PETG, as expected)

Ghostkeeper, 2020-03-26:
Yes I expanded the cutting mesh to cover the entire model. If it's only part of the first layer (certain holes but not others) then I don't think there is a way you can do this right now. You'd need to actually include those sacrificial plates in the model.

yet-another-average-joe, 2020-03-26:
Didn't think of scaling !
I won't include sacrificial plates, because it is relevant for one and only one layer height, but I will suggest users to do it by themselves.
And maybe, such a feature will be in a future Cura release (I'm unable to code this and do a pull request : the mathematics courses I hated were topology ones ; a nightmare)

I think such a feature woulkd be great for aesthetics (night and day)

Weird fact : I couldn't get the combo box with "cutting mesh" in my Cura 4.5 ; but it is in Smartavionics branch... Did you use the official release, or Smartavionics one ?

mahtDFR, 2020-03-27:
This is on the backlog as CURA-7326

fvrmr, 2021-06-04:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

LorenzGit, 2023-04-16:
Any idea on when this feature (and the other types of Sacrificial Bridging) might come to Cura?

closed 2024-04-29 by github-actions[bot]