Ultimaker/Cura #7325: [4.5] Can't Control Speed

CCS86, 2020-03-22:
I really want to take control of all posted speeds, and ditch any formulas. But I just can't seem to get there:

[image]

[image]

What am I missing?

smartavionics, 2020-03-22:
Do you have a minimum layer time set? Perhaps it's slowing the outer wall down because of that? Or maybe you have a script active that has modified the speed?

CCS86, 2020-03-22:
Doh, you called it.

That setting wasn't visible, but it was enforcing a 5 second layer time. After dropping down the perimeter count, I infringed on that. Thank you.

closed 2020-03-22 by CCS86