Ultimaker/Cura #7326: Cura 4.5 Added new material doesn't show on the material list

AESilva-IPFN, 2020-03-22:
**Application version**
Cura 4.5

**Platform**
Win10

**Printer**
Alfawise U20 One

**Reproduction steps**
1. Installed Copper3D PLActive material from Marketplace
2. Restarted Cura

**Screenshot(s)**
[image]
[image]

**Actual results**
The new material doesn't show on the material list and I can't use it (second image). But if I go to the marketplace manager it says it is installed (first image).

**Expected results**
I would expect to have the chance to choose the new material from the material list.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-03-23:
The materials on the Marketplace are targeted at Ultimaker printers, which use 2.85mm material. Since your printer uses 1.75mm material, Cura filters out the display of the materials from the Marketplace since the 2.85 mm variant would not fit in your printer anyway.

Ghostkeeper, 2020-03-23:
Duplicate of https://github.com/Ultimaker/Cura/issues/6236.

We intend to add information in the Marketplace as to which printers these material profiles are compatible with. They are indeed not showing up since your extruder accepts only 1.75mm filaments.

closed 2020-03-23 by Ghostkeeper

AESilva-IPFN, 2020-03-23:
Thank you for the information.