Ultimaker/Cura #7327: When slicing a multi-level nested model, the gaps were filled incorrectly.

gailulun, 2020-03-22:
cura4.x
windows10
**When slicing a multi-level nested model, the gaps were filled incorrectly.**

Original model
[image]

Effect of slicing
[image]
[image]
[image]

the solution
[image]
[image]
[image]
It seems that this is related to the design logic of the stl file. The added bar connects the models into a whole (three independent models before this), and also helps cura distinguish the direction of the face.

mahtDFR, 2020-03-23:
This could be down to an engine issue. Please save the project file (File -> Save...) and then zip up the .3mf file and attach to this issue so I can reproduce it.

gailulun, 2020-03-23:
[3D.zip](https://github.com/Ultimaker/Cura/files/4368656/3D.zip)
There are two models in the compressed package, one is not working properly and the other is modified.

gailulun, 2020-03-23:
Fortunately, I have n’t cleaned the recycle bin.😂

mahtDFR, 2020-03-23:
Thanks, I was able to reproduce it with the files you provided. Will discuss

rburema, 2020-03-24:
Seems to happen in master, 4.0.0 and 3.2.1 as well, so it's not due to a recent fix/regression.

rburema, 2020-03-24:
Though, if I check the model with the 'Mesh Tools' plugin (download from the Marketplace), I get the following:

[image]

We don't support models that aren't watertight or contain sub-meshes. Please adjust this in the modeller you are using.

closed 2020-03-24 by rburema

gailulun, 2020-03-24:
OK,i'll check it.