Ultimaker/Cura #7332: (Semi-)automated support generation

Sophist-UK, 2020-03-23:
This is an advanced feature request...

It seems to me that it should be possible to calculate the weight of overhanging material and (assuming some parameters about the filament being used) calculate whether supports are needed and then add them either automatically (with / without approval / adjustment by the user) or even just indicate to the user where supports might be needed.

(Also Pathio Beta allows user to simply colour in the surface of the model to show where supports are needed and then autogenerate the supports.)

fieldOfView, 2020-03-23:
See the "Generate Support" setting. This generates support automatically, based on a user-settable overhang angle. Cura also shows the areas of the model where supports sre needed in red.

mahtDFR, 2020-03-23:
[image]

@Sophist-UK As above, check support settings in Custom mode.

Since the features you describe exist already, I'll close this off.

closed 2020-03-23 by mahtDFR

Sophist-UK, 2020-03-23:
Fair enough.