Ultimaker/Cura #7333: Mimic Pathio 3D offsetting

Sophist-UK, 2020-03-23:
Feature request:

Duplicate the advanced 3D offsetting feature of Pathio.

fieldOfView, 2020-03-23:
Perhaps you could describe what it does.

fieldOfView, 2020-03-23:
See https://docs.pathio.xyz/pathio/latest/deepDives/3DOffsetting.html and https://docs.pathio.xyz/pathio/latest/features/features.html

Sophist-UK, 2020-03-23:
Thank you @fieldOfView.

Ghostkeeper, 2020-03-24:
So the difference seems to be most prominent when you're printing like an L shape. Where the vertical side of the L starts in that concave corner, Cura would produce skin all the way underneath your walls. Pathio would produce a round miter on the inside that is completely filled, leaving a bit more open for infill.

I'd say in that case, Cura's result is most likely better since this is a weak point where forces will converge in your model. You want it to produce more solid shell there than in the flats.

Is there a reason why this could be considered better for the print quality?

Sophist-UK, 2020-03-24:
As I understand it, it has 2 effects - but I am not an expert on it:

1. Suppose you have a mesh infill with a flat surface above but with a v-groove in the flat surface - Cura will (as I understand it) create a thinner surface on the sides of the v-groove i.e. the distance to the weaker mesh is small, whereas Pathio thickens the sides of the groove to improve strength and appearance.

2. As you say, it also ensures that horizontal and vertical surfaces have a neat join. I am not sure if it produces a round beaded corner or whether it thins the skin underneath causing a weakness - but even if Pathio does that, a Cura enhancement wouldn't need to do that.

Ghostkeeper, 2020-03-26:
All right, disregarding the joints then.

The sides of the V-groove are just the same as any sloped sides of the model. Cura would normally create a slope of walls there with a tiny bit of skin, like so:
[image]

As seen from the inside, indeed the joint is mitered instead of round (a bit hard to see due to the shading, but I hope it's clear):
[image]

This is similar to the "Traditional Charpy" as they depict it on their features page. The distances here are correct with the Manhattan distance of X/Y and Z. It's not too thin. The wall still has the correct width initially (red and green lines) but since the skin needs to expand underneath the walls for the mitered joint it's going to create a bit of skin there. Since this skin is not very advantageous to the strength of the model but takes a lot of time, Cura has a feature called "Skin Removal Width" that removes thin strips of skin like this. It then looks like this:
[image]

As a result, if the Skin Removal Width is too high, the joints will no longer be contributing to the thickness from a certain steep angle of the walls on. The horizontal and vertical distance still applies, but only those. The "miter joint" from combining horizontal and vertical does not contribute to the strength in these slopes. This is purely to save printing time though because in the printing process, these walls are plenty strong, stronger even than completely vertical walls.

Sophist-UK, 2020-03-26:
If Cura already has something equivalent, then please close this FR.

Ghostkeeper, 2020-03-30:
It's not really equivalent, as it fixes it in a different way, but this way we give more control over thickness in different parts. The thickness in the joints will be thicker, but I think that's generally better (and neither Cura nor Pathio really offers control about that).

closed 2020-03-30 by Ghostkeeper