Ultimaker/Cura #7335: master/4.6: Crashes when reading settings?

thopiekar, 2020-03-23:
**Application version**
Build from master (PPA)

**Platform**
Ubuntu Focal (20.04) w/ NVidia GeForce 1070 Ti

**Printer**
Doesn't matter

**Reproduction steps**
1. Remove all settings (see log)
2. Start Cura
2.1 First error section in the log is before the wizard
3. Selecting a printer, eg. UM3
3.1 Second error section follows in the log

**Screenshot(s) // Logs**
(Cura crashed.. There is nothing special to see. The log will be more interesting.)

**Actual results**
Boom!

**Expected results**
Cura just runs as usual.

**Project file**
Didn't come to the point to load anything.

**Log file**
[log.txt](https://github.com/Ultimaker/Cura/files/4371201/log.txt)

**Additional information**
Master build using Debian packaging files taken from here:
https://github.com/thopiekar/Cura-packaging

fieldOfView, 2020-03-24:
This "smells like" it is related to the merging of https://github.com/Ultimaker/Uranium/pull/587

Ghostkeeper, 2020-03-24:
Not getting this in my development environment by the way, on Ubuntu 19.10.

Maybe it's related to that. It did alter things in the SettingFunction class after all. But I wonder what line it's actually crashing on that tries to call `startswith` on an integer or something. Maybe, like, this one?

https://github.com/Ultimaker/Uranium/blob/a4781002ceaad051bce065d4d766cf1cee2e78c9/UM/Settings/SettingFunction.py#L239

fieldOfView, 2020-03-24:
More likely this, a couple of lines lower:
https://github.com/Ultimaker/Uranium/blob/a4781002ceaad051bce065d4d766cf1cee2e78c9/UM/Settings/SettingFunction.py#L257
That will only be used with Python 3.8+ (according to the comments). Apparently visit_Constants is also called for non-string constants (in Python 3.8+)

fieldOfView, 2020-03-24:
@thopiekar could you test if changing line 257 to this fixes the issue?
`if isinstance(node.value, str) and node.s.startswith("_"):`

fieldOfView, 2020-03-24:
See https://github.com/Ultimaker/Uranium/pull/592

closed 2020-03-25 by nallath