Ultimaker/Cura #7336: Drop-down menus missing in 4.4.1 and 4.5

juren53, 2020-03-23:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-03-24:
If it's really the shader cache, then there's a more permanent workaround implemented in 4.5

If you edit `cura.cfg`, put `force_empty_shader_cache = True` under the `[view]` header (if there isn't a `[view]` header, you can create one).

Ghostkeeper, 2020-03-25:
Duplicate of https://github.com/Ultimaker/Cura/issues/5296.

closed 2020-03-25 by Ghostkeeper