Ultimaker/Cura #7337: Cura 4.5 Won't Launch Either

bxlaster0611, 2020-03-23:
**Application version**

Cura 4.5

**Platform**

Windows 10 (latest updates as of 3/23/20)

**Printer**
Ender 3

**Reproduction steps**

Launch Cura then nothing happens

**Screenshot(s)**

[image]

[image]

**Actual results**

I get the blue circle hourglass that then disappears; nothing else after

**Expected results**

Cura should launch lol

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**

All the same crash details can be found at the following link from the same issue happening before for me. The only way I got it to work was to factory reset my laptop. Now after installing windows updates, cura no longer launches. Please see my attached crash screenshot (basically same as before) and the response I got via email from Cura support.

https://github.com/Ultimaker/Cura/issues/7035

mahtDFR, 2020-03-25:
One of the causes of this in previous versions was outdated Intel drivers (see discussion #6197)
Are you using Intel on your machine? If so, try updating the drivers.
Were any logs generated? If so, can you attach here? We need more info before we can provide useful help.

mahtDFR, 2020-03-26:
@bxlaster0611 just following up - any progress on this?

bxlaster0611, 2020-03-27:
I am indeed running an Intel core i7 vPron 7th Gen Processor. I am in the process of updating my drivers through “Driver Easy” but not sure this will work. What do you mean by logs? I have a diagnostic report of the crash Sent from Mail for Windows 10 From: mahtSent: Thursday, March 26, 2020 9:25 AMTo: Ultimaker/CuraCc: bxlaster0611; MentionSubject: Re: [Ultimaker/Cura] Cura 4.5 Won't Launch Either (#7337) @bxlaster0611 just following up - any progress on this?

mahtDFR, 2020-03-27:
> What do you mean by logs?

https://github.com/Ultimaker/Cura#logging-issues

closed 2020-06-19 by no-response[bot]