Ultimaker/Cura #7340: [4.5.0]Cura crashes when using multiple displays (screens)

Sahrio, 2020-03-24:
**Application version**
Ultimaker Cura 4.5.0

**Platform**
Win10
GPU: GeforceGTX970m
intel i7 6700HQ
16gbRAM
Two displays

**Printer**
Anet A8
Selected Custom FFF printer with the settings of an ANET A8

**Reproduction steps**
1. With the secondary display disconnected launch the Cura application.
2. Now connect the secondary display.
3.Drag the Cura window from main display to secondary display.
4.Close the Cura window on the secondary display.
5.Relaunch Cura

**Screenshot(s)**
[image]
Cura launches correctly on the main display

[image]
Dragging cura to the secondary display

[image]
Closing Cura in the secondary display

[image]
Launching Cura again

[image]
Cura crashes and never loads. Indeed, its somehow hard to get the "end process question" from windows to terminate the crashed app without going to the Task Manager.

**Actual results**
Cura crashes and stays blank

**Expected results**
Cura launches correctly on the secondary display.

**Project file**
Doesn´t matter. This happens even with no files being loaded.

**Log file**
(https://github.com/Ultimaker/Cura/files/4374692/cura.log)

**Additional information**
My main display and my secondary display dont have the same resolution nor ratio. I suppose the problem could be the different ratios, one is 16:9 and the other is 4:3.

fieldOfView, 2020-03-24:
As a workaround, in the General preferences there's an option "Restore window position on start". If you uncheck that, Cura will always start on the primary screen.

Ghostkeeper, 2020-03-25:
Duplicate of https://github.com/Ultimaker/Cura/issues/5651

closed 2020-03-25 by Ghostkeeper