Ultimaker/Cura #7342: Cura 4.5 Setting Extruder Temp Above 365C

StrikerTC, 2020-03-24:
**Application version**
Cura 4.5

**Platform**
Windows 10

**Printer**
FUNMAT HT

**Reproduction steps**
I operate a high temperature printer which requires extruder temperatures 380C to 420C. In previous versions of Cura (just upgraded from 4.41), this hasn't been an issue. Cura will give me a warning (orange box) but that's it. Now in 4.5, Cura gives me a red box at any printing temperature over 365C and wont let me slice the model. This happens with any profile I pick.

**Screenshot(s)**
[image]

**Additional information**
Is there a config setting somewhere that sets a max printing temperature?

rburema, 2020-03-24:
Which printer-profile do you use with that printer? Is it just the Custom/FFF one?

fieldOfView, 2020-03-24:
See this thread on the Ultimaker community forum for a fix:
https://community.ultimaker.com/topic/31565-peek-400%C2%B0c-not-able-to-slice/?tab=comments#comment-259028

StrikerTC, 2020-03-24:
> Which printer-definition do you use with that printer? Is it just the Custom/FFF one?

I modified the Custom/FFF and renamed it.

rburema, 2020-03-25:
@StrikerTC It's all explained in the linked thread above.

In the long term, it's probably best to get a separate printer definition from either the manufacturer or someone in the community. If someone has a 'good' one (we wont be able to test, since we don't have much in the way of 3rd party printers, even less so since we all work from home now), we might be able to merge it.

StrikerTC, 2020-03-25:
> @StrikerTC It's all explained in the linked thread above.
>
> In the long term, it's probably best to get a separate printer definition from either the manufacturer or someone in the community. If [...]

Thanks for the help. I'll see if I can get a hold of a proper definition file. Until then, I'll just go back to using 4.41. I'm just a little frustrated the this was changed in the latest release with no easy way to change the value to what I need. It would be nice if there was a place to set max temp values when setting up a custom printer. It also doesn't help that my work laptop is locked down to where I can't edit the definition files myself.

rburema, 2020-03-25:
>  It also doesn't help that my work laptop is locked down to where I can't edit the definition files myself.

Cura also accepts files in the user directory, `%appdata%/cura/4.5/...` maybe that'll help?

GregValiant, 2024-12-15:
There are a couple of different workarounds for this.  I have a feeling that the attorneys for the underwriters insisted on the limits for the hot end and bed as far-and-away the greater number of printers cannot handle high temperature printing.
I'll go ahead and close this.

closed 2024-12-15 by GregValiant