Ultimaker/Cura #7343: when printing multiple identical models at one time (similar to an array), cura has a tolerable threshold.

gailulun, 2020-03-24:
when printing multiple identical models at one time (similar to an array), cura has a tolerable threshold. When the number increases to a certain value, you can still slice at this time, and you cannot slice even if you increase one.At this point, the slicing progress bar will stop at 0% and occupy a lot of memory. This threshold looks very low.This seems to have nothing to do with the performance of the computer.For computers with smaller memory, it seems that only the slice time is affected.This problem does not exist in other slicing software.

rburema, 2020-03-24:
Could you maybe attach a project file where this happens? (Save, don't export, and compress it via zip to get github to accept the attached file.)

gailulun, 2020-03-25:
The sliceable threshold of this model is 35, and it cannot be sliced ​​when it is increased to 36 (the CPU usage is almost zero when sliced at this time). And when saving as a .3mf file, cura almost used all my 24GB memory and 100% hard disk IO. The CPU utilization also soared, but the progress bar hardly moved to save it. I waited for a minute to save the 35 models.But it was too big to upload.This threshold should also be related to the complexity or size of the model. When I replaced the model with a more complex model(Including more rounded corners), even 20 of them could not be sliced ​​at this time.

[models.zip](https://github.com/Ultimaker/Cura/files/4380298/models.zip)

gailulun, 2020-03-25:
sorry it was too late in China then(1:00 AM Beijing Time by then.) and I just saw the news.

smartavionics, 2020-03-25:
Hi, please provide a project file (do File->Save) and zip and attach the .3mf file. We need to see that so we know the settings you are using. Thanks.

smartavionics, 2020-03-25:
Just save the project file with no models on the buildplate.

gailulun, 2020-03-25:
> Hi, please provide a project file (do File->Save) and zip and attach the .3mf file. We need to see that so we know the settings you are using. Thanks.

it's almost 170MB

smartavionics, 2020-03-25:
Even without the models ?

gailulun, 2020-03-25:
no models?

gailulun, 2020-03-25:
like this？
[image]
😂

smartavionics, 2020-03-25:
Don't worry, I have found what the problem is. Devs, this is what I get when I try to slice 40 parts...
```
2020-03-25 10:55:30,138 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [444]: Sending slice message took 3.830373764038086 seconds
2020-03-25 10:55:31,836 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] [libprotobuf ERROR /home/markb/3dp/cura/cura-build-environment.git/build/Protobuf-prefix/src/Protobuf/src/google/protobuf/io/coded_stream.cc:207] A protocol message was rejected because it was too big (more than 524288000 bytes).  To increase the limit (or to disable these warnings), see CodedInputStream::SetTotalBytesLimit() in google/protobuf/io/coded_stream.h.
2020-03-25 10:55:31,836 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] [ERROR] Failed to parse message:
2020-03-25 10:55:31,837 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] ðç
2020-03-25 10:55:31,838 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] ©ÌØ´àÌffÀ=
2020-03-25 10:55:31,838 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] ÿÀÀÌÌ=ffÀÁ@áú>ffÀÁ

```

smartavionics, 2020-03-25:
I think the only workaround would be to reduce the number of polygons in the model or print less than 36 models in one go.

gailulun, 2020-03-25:
I think so too.

gailulun, 2020-03-25:
Does this mean that if a single model exceeds 500M, it cannot be sliced.

gailulun, 2020-03-25:
perhaps. . . We can partition the slices, reduce the amount of data contained in each zone to less than 500MB, and integrate the code into a file after the slice is completed. That will be a big project. . .

rburema, 2020-03-25:
Thanks for the analysis @smartavionics 
I'll bring it up in the next discussion with the (other) devs.

@gailulun That might be a good approach to fixing it, yes. Not sure if that'll actually be a big project (it may be), that depends on the details. In any case, seems for now you should use the workaround, until we've fixed this (which might not be for a while, depending on priorities).

gailulun, 2020-03-25:
Very happy to contribute to you😀

Ghostkeeper, 2020-03-25:
Duplicate of https://github.com/Ultimaker/Cura/issues/3105. There's a lot of analysis in that topic too. Please refer there.

closed 2020-03-25 by Ghostkeeper