Ultimaker/Cura #7347: Cura 4.5 will not load any material

dpservis, 2020-03-25:
**Application version**
4.5

**Platform**
Windows 10

**Printer**
Ultimaker 2

**Reproduction steps**
1. Install Cura
2. Open Cura

**Screenshot(s)**
[image]

**Actual results**
No materials are available (only one I imported manually)

**Expected results**
All materials should be available

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
`2020-03-23 13:10:02,999 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FileLogger
2020-03-23 13:10:03,027 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FirmwareUpdateChecker
2020-03-23 13:10:03,042 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FirmwareUpdater
2020-03-23 13:10:03,055 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeGzReader
2020-03-23 13:10:03,070 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeGzWriter
2020-03-23 13:10:03,088 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeProfileReader
2020-03-23 13:10:03,116 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeReader
2020-03-23 13:10:03,132 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeWriter
2020-03-23 13:10:03,152 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin ImageReader
2020-03-23 13:10:03,168 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin LegacyProfileReader
2020-03-23 13:10:03,182 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin LocalContainerProvider
2020-03-23 13:10:03,203 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin LocalFileOutputDevice
2020-03-23 13:10:03,217 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin MachineSettingsAction
2020-03-23 13:10:03,231 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin ModelChecker
2020-03-23 13:10:03,248 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin MonitorStage
2020-03-23 13:10:03,268 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin PerObjectSettingsTool
2020-03-23 13:10:03,283 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin PostProcessingPlugin
2020-03-23 13:10:03,296 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin PrepareStage
2020-03-23 13:10:03,311 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin PreviewStage
2020-03-23 13:10:03,333 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin RemovableDriveOutputDevice
2020-03-23 13:10:03,345 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SentryLogger
2020-03-23 13:10:03,379 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SimulationView
2020-03-23 13:10:03,396 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SliceInfoPlugin
2020-03-23 13:10:03,410 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SolidView
2020-03-23 13:10:03,425 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SupportEraser
2020-03-23 13:10:03,498 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin Toolbox
2020-03-23 13:10:03,513 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CameraTool
2020-03-23 13:10:03,531 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin MirrorTool
2020-03-23 13:10:03,552 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin RotateTool
2020-03-23 13:10:03,574 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin ScaleTool
2020-03-23 13:10:03,587 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SelectionTool
2020-03-23 13:10:03,607 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin TranslateTool
2020-03-23 13:10:03,620 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin TrimeshReader
2020-03-23 13:10:03,643 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin UFPReader
2020-03-23 13:10:03,661 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin UFPWriter
2020-03-23 13:10:03,679 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin UltimakerMachineActions
2020-03-23 13:10:03,898 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin UM3NetworkPrinting
2020-03-23 13:10:03,916 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [39]: Checking for new version of cura
2020-03-23 13:10:03,920 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin UpdateChecker
2020-03-23 13:10:04,328 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin USBPrinting
2020-03-23 13:10:04,541 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin VersionUpgrade21to22
2020-03-23 13:10:04,616 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin VersionUpgrade22to24
2020-03-23 13:10:04,706 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin VersionUpgrade25to26
2020-03-23 13:10:04,793 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin VersionUpgrade26to27
[... 428 lines cut ...]
2020-03-23 13:31:56,224 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Renderer: AMD Radeon R9 200 / HD 7900 Series
2020-03-23 13:31:56,227 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: GLSL Version:    4.60.0
2020-03-23 13:31:56,231 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\default.shader]...
2020-03-23 13:31:56,237 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\selection.shader]...
2020-03-23 13:31:56,242 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\select_face.shader]...
2020-03-23 13:31:56,247 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\default.shader]...
2020-03-23 13:31:56,252 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\composite.shader]...
2020-03-23 13:31:56,258 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\overhang.shader]...
2020-03-23 13:31:56,264 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\striped.shader]...
2020-03-23 13:31:56,271 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\transparent_object.shader]...
2020-03-23 13:31:56,277 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\striped.shader]...
2020-03-23 13:31:56,282 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\default.shader]...
2020-03-23 13:31:56,288 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\grid.shader]...
2020-03-23 13:31:56,295 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\platform.shader]...
2020-03-23 13:31:56,686 - WARNING - [MainThread] cura.Machines.Models.IntentModel._getActiveMaterials [119]: Could not find the material FormFutura_ASAX #3
2020-03-23 13:31:56,691 - WARNING - [MainThread] cura.Machines.Models.IntentModel._getActiveMaterials [119]: Could not find the material FormFutura_ASAX #3
2020-03-23 13:31:56,698 - WARNING - [MainThread] cura.Machines.Models.IntentModel._getActiveMaterials [119]: Could not find the material FormFutura_ASAX #3
2020-03-23 13:31:56,702 - WARNING - [MainThread] cura.Machines.Models.IntentModel._getActiveMaterials [119]: Could not find the material FormFutura_ASAX #3
2020-03-23 13:31:57,658 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [516c165e6ece4b3db0484fd449373f26] finished.
2020-03-23 13:31:57,662 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-03-23 13:32:00,218 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [187]: Error check finished, result = False, time = 11.4s
2020-03-23 13:32:07,632 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles`

**Additional information**
I installed Cura 4.5 upgrading from 4.3. No materials are available when I start 4.5. The material files seem to be copied in the Roaming forlder but they are not read...

konskarm, 2020-03-26:
I did a fresh install of Cura 4.3, 4.4.1, and 4.5. None of the versions allows you to select the material in an Ultimaker2. Not sure why the option appears for you, since Ultimaker 2 never allowed for that.
Where did you find the DuraPro ASA material? I cannot locate it in the Marketplace.

dpservis, 2020-03-26:
> I cannot reproduce the issue. I did a fresh install of Cura 4.3, 4.4.1, and 4.5. None of the versions allows you to select the material in an Ultimaker 2. I am not sure why the option appears for you, since Ultimaker 2 never allowed for that.
> 
> Where did you find the DuraPro ASA material? I cannot locate it in the Marketplace.

My Ultimaker is attached to Octoprint via the Cura plugin. 

The DuraPro ASA material I made it myself on Cura 4.3. I have defined several materials that I use often on Cura 4.3 but none of them is automatically imported. The DuraPro ASA I imported it manually which is why it is the only one available. But not even the Cura generic materials are available.

Ghostkeeper, 2020-03-26:
The Ultimaker 2 will allow materials if you change the g-code flavour to RepRap.

It would help to reproduce this if you can save a project file and/or share that DuraPro ASA you created.

dpservis, 2020-03-26:
> The Ultimaker 2 will allow materials if you change the g-code flavour to RepRap.
> 
> It would help to reproduce this if you can save a project file and/or share that DuraPro ASA you created.

As I am using Octoprint, I have changed to Marlin flavor. I will attach projects and material asap.

dpservis, 2020-03-26:
Files uploaded
[Cura 4.5 bug.zip](https://github.com/Ultimaker/Cura/files/4389974/Cura.4.5.bug.zip)

Ghostkeeper, 2020-03-30:
The problem indeed reproduces by loading that 4.3 project file in, and then upgrading to 4.5. Strangely you can select materials fine if you add a new Ultimaker 2 and then change the g-code flavour.

Looking at the machine instance in 4.5, it seems that the version upgrade thinks that you've installed the Olsson block since, after changing the g-code flavour, your printer now has nozzle profiles. The Olsson block upgrade allows nozzle profiles, but not material profiles. The problem is that the machine_instance is exactly the same if you change the g-code flavour vs. if you check the Olsson block printer, so it's hard to make it upgrade correctly in this case. I'll ask the other developers if they have an idea what we should do with this.

dpservis, 2020-03-30:
Thanks for the feedback. I have installed the Olsson block and registered the change in 4.3 as well. I have also installed the Bondtech extruder so my U2 is further away from the original...

GrunTed, 2020-04-01:
Hi All, I also cannot see or change material types for UM2 in latest Cura.
Apologies, I know nothing about coding etc, but want to sort this out, you mentioned changing the G-Code flavour fixes..... is changing the flavour something a noob like me can do? can you give me an idiots guide please?

Ghostkeeper, 2020-04-01:
Yeah it's not hard. You click on the "Manage Printers" button in the drop-down here:
[image]

Then click on "Machine settings" here:
[image]

Then there's this drop-down to select the g-code flavour here:
[image]

I've printed with the RepRap flavour for tests and it seemed to work fine. You're changing it away from Ultimaker's defaults so a couple of things might change. I'd expect that the printer might take a while longer to heat up at the start of the print. I don't expect the print quality to be affected though unless you change the defaults for temperature, retraction or fan speeds.

Ghostkeeper, 2020-04-01:
We've discussed this issue with the team. We've decided to defer this. It won't get any priority from us, but it's a valid bug so a fix is welcome. Several things played into this decision:

* It only occurs with a modification that not a lot of users made.
* There is a workaround: Create a new printer and change the g-code flavour again. To transfer any settings that were not saved to a profile you'd need to store them in a profile (or just write them down and apply them again on the new printer).
* The bug is in the version upgrade from 4.3 to 4.4. We can fix it, but the fix will only be applied when opening a newer version of Cura than 4.3 for the first time. Existing users of 4.4 or 4.5 that have this issue won't get the fix applied to their configuration. Their configuration is basically corrupt. Though don't worry, I just advise you to delete that old printer from Cura and re-add it.

GrunTed, 2020-04-01:
Thanks for the help!

fieldOfView, 2020-04-01:
> I've printed with the RepRap flavour for tests

The Marlin flavor is actually more correct for the Ultimaker 2 family; the RepRap flavor is meant for machine using reprap firmware (eg machines using the Duet family of boards).

rpadams, 2020-04-05:
I'm also seeing this.  Fresh install of Cura 4.5 on Mac (deleted old pref directory, etc.) does not allow you to add/modify materials for the Ultimaker 2 without changing the g-code flavor.  I changed it to Marlin and suddenly I had materials again.

fieldOfView, 2020-04-05:
You are seeing this because it is by design. 

The idea of UltiGcode on the Ultimaker 2 was to explicitly not select the material on the slicer, so you could slice once and print with any material by selecting the material on the printer. Since then, lots more materials came to market, and Ultimaker changed their mind about selecting materials only on the printer for newer models. However, the Ultimaker 2 behavior was not changed. At least you have the option to make the change in Cura. It is however not a bug, it is intended behavior.

rpadams, 2020-04-05:
In all previous versions of Cura I have selected the material properties in g-code, so it's hard not to read this response as "We changed something in software so that when you click the software update button it silently makes your $2,500 machine unusable without modifying the preferences in an undocumented way."

fieldOfView, 2020-04-05:
Then apparently, in the other versions of Cura, you had the gcode flavor not set to "Ultimaker 2". 

Is there a problem with the workaround of setting the gcode flavor to "Marlin"?

This is not something that was changed in Cura 4.5.

rpadams, 2020-04-05:
No, no problem changing the setting other than the hours I wasted trying to find out why the upgrade to Cura 4.5 broke everything, ultimately requiring me to find my way to this Github issue to resolve it. (Add this to my list of reasons that I regret buying an Ultimaker.)

fieldOfView, 2020-04-05:
I am sorry that happened to you. I am just an external contributor who decided to answer your question on a sunday night.

rpadams, 2020-04-05:
I appreciate your response and your contributions.

dpservis, 2020-04-06:
My findings so far: 

1. If I introduce a new Ultimaker 2 printer with Marlin, I have all materials and no nozzles
2. If I introduce a new Ultimaker 2 with Olsson block and Marlin, I have no materials and all nozzles

Not many users introduced the Olsson block? I would be surprised if any didn't

Ghostkeeper, 2020-04-08:
I think we have a different bug there @dpservis, though it's very much in the same ballpark.

The Olsson block should add only swappable nozzles to your printer. It doesn't normally add the ability to let Cura decide on temperatures. That's what changing the g-code flavour does in this case.
When you swap from "Ultimaker 2" to "Marlin" when using an Ultimaker 2, it then allows you to select materials. However when you do the same when using an Ultimaker 2 with Olsson block, the result is different. It doesn't auto-select PLA, it has no available materials, and when you restart you suddenly only get custom materials which is extra weird.

> Not many users introduced the Olsson block? I would be surprised if any didn't

A few percent of the UM2 users have it according to Cura's anonymous statistics. I can't see right now how many are using the unsupported Marlin g-code flavour though.

dpservis, 2020-04-16:
I didn't manage to figure out a workaround so far... (didn't put much effort though I was super busy and printing on top). So to recap:

Preconditions:
1. Ultimaker 2 with 3dSolex hotend and Bondtech extruder
2. Cura 4.3 where I defined Ultimaker 2 with Olsson block
3. Octoprint plugin
4. Marlin flavor
5. Nozzles and materials available

STR
1. Install Cura 4.5 without uninstalling Cura 4.3
2. Open Cura 4.5
3. Create new Ultimaker 2 with Olsson block printer 
4. Define Marlin flavor

ER: All materials and nozzles available
AR: Only nozzles available

fieldOfView, 2020-04-16:
> 1. Ultimaker 2 with 3dSolex hotend and Bondtech extruder
> 2. Cura 4.3 where I defined Ultimaker 2 with Olsson block

Try adding an Ultimaker 2+ to Cura instead. Your upgraded Ultimaker 2 resembles the Ultimaker 2+ more than it does an Ultimaker 2.

dpservis, 2020-04-16:
Good idea, I will try it

Ghostkeeper, 2020-04-22:
We've made a fix for updating the nozzles in this case so that second bug is fixed now. The upgrade is not fixed yet.

hpandelo, 2020-04-25:
Same occurring to me on v4.6. It works fine when I set the material type as ABS, but if I type ASA the profiles won't load

dpservis, 2020-04-27:
With Ultimaker 2+ I have nozzles, I have materials but I don't have profiles now. 

Why is this so complicated? I mean, some flexibility could be beneficial for the software...

Ghostkeeper, 2020-04-28:
> Same occurring to me on v4.6. It works fine when I set the material type as ABS, but if I type ASA the profiles won't load

This is expected. The quality profiles are material-specific, so if you change the material type there are no quality profiles for your material any more. You'd have to create those profiles yourself.

> Why is this so complicated? I mean, some flexibility could be beneficial for the software...

It is so complicated because Cura has an enormous amount of flexibility to accommodate all types of printers, honestly. Your materials might not have been appearing because the extruder had the wrong compatible material diameter set. Your quality profiles might not be appearing because there are no specialised quality profiles for your material+nozzle+printer combination and so this combination is unsupported and you'll have to create your own quality profiles.

If you provide project files we might be able to help out to see what's wrong.

hpandelo, 2020-04-28:
> This is expected. The quality profiles are material-specific, so if you change the material type there are no quality profiles for your material any more. You'd have to create those profiles [...]

1. ASA can use the same profile as ABS. What's the reason to don't have it?

2. I believe the main problem occurring into all parts of Cura is the "magic" stuffs done in background with no User interaction. For example, OK if we have no profiles, but add a message "No profiles available due to .......". No problem to have some "rules" into the software, but the user should be notified to know what should be done to fix the "issue"

**Another examples:**

* Yellow fields, it just appears when you set some values, but what it means? Where I can find what information the field is trying to pass to me being yellow?

* Value overriding, ok to have values being overridden into the material/print settings (I still doesn't agreeing with this behavior haha 🦊), but it should be explicit to the user and should have a way to reset it when needed. For example, I have a profile saved with temperature X and if I change the material it doesn't affect and I have no way to change it back to original behavior.

I believe the UX in Cura should be reviewed by someone with expertise into this area

Ghostkeeper, 2020-04-29:
> ASA can use the same profile as ABS. What's the reason to don't have it?

Cura doesn't know what ASA or ABS is. Internally the rule we implemented is that materials with the same "Material Type" will share their quality profiles. That way you can have different materials with the same material type and don't need to duplicate the quality profiles a million times for each possible colour of Ultimaker PLA for instance.

> I believe the main problem occurring into all parts of Cura is the "magic" stuffs done in background with no User interaction. For example, OK if we have no profiles, but add a message "No profiles [...]

Point is that for many cases this would just not be an issue. Yes, the user made their own type of plastic and adds a profile for it in Cura, of course there will not be any profiles for it.

> Yellow fields, it just appears when you set some values, but what it means? Where I can find what information the field is trying to pass to me being yellow?

We intend for the orange/yellow colour to indicate a warning. Something that could well be not intended by the user or something we think is not going to print well. This is indeed only "intuitive" and not explicitly explained in the Cura manual. I think these orange and red colours are quite intuitive though and I wouldn't know where to put this explanation explicitly in Cura's interface if that's what you are requesting.

> Value overriding, ok to have values being overridden into the material/print settings (I still doesn't agreeing with this behavior haha 🦊), but it should be explicit to the user and should have a [...]

Yes this has been requested a number of times. We've implemented this before in what is now these little fx buttons but the behaviour is less than ideal and the clarity of this feature even less. This is still in our backlog.

> I believe the UX in Cura should be reviewed by someone with expertise into this area

We have UXers with formal training in our team (not me though, don't ask me). And also formerly in our team. They make all designs for our interface changes, including the aforementioned examples, although they are sometimes more of an arbiter to judge if an externally contributed design is okay (like here in a PR). The problem is a bit that it seems to be inherently subjective in some areas. And there is always the big conflict between expert users that want more knobs and control and novice users that would do better with fewer knobs and stuff done automatically for them.

closed 2023-06-24 by github-actions[bot]