Ultimaker/Cura #7348: Cura 4.5.0 does not recognise the printer via USB

dxc-vv, 2020-03-25:
**Application version**
Cura 4.5.0

**Platform**
Windows 10

**Printer**
Anycubic Kossel Linear Plus

**Reproduction steps**
1. Turn On PC
2. Turn On Printer
3. Connect the printer to the PC with USB cable
4. Open Cura 4.5.0
5. Open Tab "Control"
6. Cura Show some functionality to control the printer but they doesn't work

**Actual results**
I'm not able lo send any command to print via USB
I'm not able to print via USB Cable

**Log file**
attached

**Additional information**
I tryed to update the printer driver but nothing chage.
Cura 4.0.0 find the printer properly.
[cura.log](https://github.com/Ultimaker/Cura/files/4381188/cura.log)

mahtDFR, 2020-03-25:
Your log shows this:

`2020-03-25 09:28:20,187 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [80]: Detected baud rate 250000 on serial COM5 on retry 0 with after 0.02 seconds.`

It seems to be connecting to something. Is there another device occupying a COM port, e.g. an Arduino? If so try removing anything else to see if you can connect.

dxc-vv, 2020-03-25:
The only thing connected to the PC besides the printer is the mouse. 
The strange thing is that Cura 4.0.0 finds the printer without problems ...

mahtDFR, 2020-03-26:
Can you share a log output from 4.0 when it works?

Ghostkeeper, 2020-03-26:
> Your log shows this:
>
> `2020-03-25 09:28:20,187 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [80]: Detected baud rate 250000 on serial COM5 on retry 0 with after 0.02 seconds.`
>
> It seems to [...]

It only shows that if it got a valid temperature response from a temperature query via USB. It's a pretty specific response. It must almost be a printer in COM5 then.

eskeyaar, 2020-04-28:
maybe it needs to connect at different baudrate?

Ghostkeeper, 2020-04-29:
The log file indicates that it "detected baud rate 250000". It says that only when it got a response with the correct format for a temperature probe `M105`, so the baud rate must be correct then, unless it's like random noise.

eskeyaar, 2020-04-29:
well, sometimes my machine gets detected at 2 different baud rates 115200 and 250K. but only 250K was able to send and receive commands via repetier host, i thought in that way.

Ghostkeeper, 2020-04-29:
Oh, well that could be then. I'd think that if the baud rate is halved you'd get duplicate bits and the characters would change completely though. But I must admit that I'm not that well versed in firmware/electronics programming to know how that sort of thing works further than my own intuition.

dxc-vv, 2020-04-29:
Hi , thanks for suggestions, now cura work fine, i have re-installed the software and the drivers, flashed Marlin 2.0 e now the printer is recognized as well.
Many thanks to all.

eskeyaar, 2020-04-29:
> 
> 
> Hi , thanks for suggestions, now cura work fine, i have re-installed the software and the drivers, flashed Marlin 2.0 e now the printer is recognized as well.
> Many thanks to all.

What was the previous firmware? i just wanted to solve issue am facing in another machine. maybe it's the firmware problem for me too..

dxc-vv, 2020-04-29:
The previous firmware was the original one... i never flashad other firmware before Marlin 2.0.

closed 2020-04-30 by Ghostkeeper