Ultimaker/Cura #7350: [BUG] No brim no support on the inside of models

army2500, 2020-03-25:
Brim settings:
ON
Only on outside of model
Replaces support

Support settings the usual.

The result on initial layer was that there was no brim and no supports on the inside of model.

Disabling brim setting "only on outside of model"
enabled back support on initial layer on the inside.

But i think that if you disable brim on the inside, than supports on the inside should not be effected.

mahtDFR, 2020-03-26:
Thanks for the report. Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so we can try to reproduce it.

If you haven't already, please can you attach a [log file](https://github.com/Ultimaker/Cura#logging-issues).

army2500, 2020-03-26:
Hello!

Sorry i am completely new to Cura and Bug reports. I currently do not have
an option to reproduce, as i have models and cura on another computer.
I am attaching a screen capture of the first layer. The supports of the
first layer were omitted almost completely, apart from 2 small blobs.
On the second layer supports were normal, but this caused the supports to
not stick to print bed.
I presume this happened because i have ticked the box "Brim replaces
support".
but at the same time a had also ticked the box "brim only on outer edge" so
supports on the inside should not be affected by my opinion.
Brim settings were: 5mm from model and 3mm tick i believe.
[image: image.png]
Sincerely, Rok

mahtDFR, 2020-03-26:
Why do you want to use that setting? For starters try disabling it.

I can't see a screen capture attached. Let me know when you can provide a project file, then we can try to help you more.

Ghostkeeper, 2020-03-30:
To add attachments, you'd need to go to your issue in the github.com website: https://github.com/Ultimaker/Cura/issues/7350 Post it there. If you reply to e-mail notifications the attachments don't get posted.

closed 2020-06-19 by no-response[bot]