Ultimaker/Cura #7352: CURA 4.5.0 -  Filling protruding out from the shell

Scorpiaire, 2020-03-26:
Hello.

After several tests on similar STLs, the filling crosses the shell on the free side of the material (in Orange on the attached image)

[image]

smartavionics, 2020-03-26:
Please provide the project file (File -> Save) and then rename or zip the .3mf file and attach to this issue. thanks.

Scorpiaire, 2020-03-26:
Sorry it's a 27Mo Zip...
I shared you the dropBox Link

https://www.dropbox.com/s/qrleshtyw1k202d/U19-026%20141%2001-A30.zip?dl=0

Thanks

smartavionics, 2020-03-26:
Thanks for the project file. Cura mesh tools says that one of the objects is not watertight, fix the model and it should print OK.

[image]

Scorpiaire, 2020-04-16:
Sorry guys I am comming back with this topic !
I apology for my english which is not the best we can see.

I agree in my previous example , the stl I wanted to print were not waterthight, and maybe explain why it goes wrong.

So i tryed again because I don't stop working in spite of the Covid 19 confinement in France.
I meet again this bug of Filling protruding out from the shell. and I investigate.
I took a stl I already printed without problem with Cura 4.4. And I meet the Bug angain with the v4.5.0 whereas in the v4.4 it won't...

Neverthelesss I found something interesting withe the v4.5.0.
In all the cases witch are going wrong , the Gradual infilling is activated. When I desactivate the gradual infilling , this matter disappear.
In all probability the bug came from this option.

[image]
[image]

I share you the dropbox file where you can find the .3mf and the .slt files.
https://www.dropbox.com/sh/9qwzg6jdopcc59w/AADjR4by7zeMwjQaJdEMge9ja?dl=0

Thank you for your help

Ghostkeeper, 2020-04-22:
I also don't see anything wrong with your model, indeed. However I'm unable to reproduce the problem with those files.
[image]

However I'm clearly using very different settings than you, as my printing time is more than twice as long. Could you share a project file? Project files are more than just 3MF files, since they also contain information on which printer you're using and which settings you've changed. It should allow us to reproduce the slice pretty much exactly (as long as we're using the same version). You can save a project file through File -> Save...

Scorpiaire, 2020-04-22:
Thank You for reopening this topic ;)

Find with the dropbox link Below 2 Project Files .3MF saved as you asked me.
Corresponding to our printers. Alfawise U20+ and Creality CR10 S5
In both of them the problem exist.

https://www.dropbox.com/sh/pvyw7hyr16q7qy8/AAA1wrLMek4Tvt4GrsYRJHh5a?dl=0

For information.
1) I use for both machine a custom FFF printer , because if I don't use a custom Printer I can't modify the nozzle diameter.

2)last week we instal Cura 4.5.0 on a collegue computer , we import my profile on his machine , we created the printers from nothing with the custom FFF printer.
and he also met this problem.

I just saw that 4.6.0 is now available , maybe the problem does not exist with this version? must try...

Thanks for your help.

Ghostkeeper, 2020-04-28:
It seems that this is caused by the "Infill Support" feature. That one seems to not work together nicely with gradual infill. Disabling "Infill Support" in your case fixes the problem at no additional print time cost.

With an infill overhang angle of 5 degrees the effect of this is pretty minute anyway. There would be some models with very steep 5-degree-angle walls where you could save on a bit of infill, but the savings there are so small then because there is not a lot of area underneath those steep walls. You should consider removing that from your profile.

I'll discuss this issue with my colleagues though for the general case.

Ghostkeeper, 2020-06-29:
Initially this was added to our backlog. However considering new issues that came up and considering how uncommon this combination of settings is, we've decided to defer it for now.

We won't get to fixing it soon, so for expectation management we need to indicate that we let this one slip until we have more time or it becomes a more common problem.

closed 2023-06-24 by github-actions[bot]