Ultimaker/Cura #7353: I cannot change the display language,even though I have restart Cura.

civic220, 2020-03-26:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-03-26:
The issue template you've posted isn't very helpful here. Have you changed the display language in settings?

fieldOfView, 2020-03-26:
How do you run Cura, and on what platform? It may take additional steps to get the translations working if you run Cura from source.

civic220, 2020-03-26:
> The issue template you've posted isn't very helpful here. Have you changed the display language in settings?

> The issue template you've posted isn't very helpful here. Have you changed the display language in settings?

thanks for reminding me. 

> How do you run Cura, and on what platform? It may take additional steps to get the translations working if you run Cura from source.

I run cura from source, use the command line "python cura_app.py"…The platform is windows10.

civic220, 2020-03-26:
> The issue template you've posted isn't very helpful here. Have you changed the display language in settings?

I havw changed the display language in settings.However，it doesn't work.

fieldOfView, 2020-03-26:
It would be my guess that your resources/i18n folder only contains .po files, and not the "compiled" .mo files. Those can be found here:
https://github.com/Ultimaker/cura-binary-data/tree/master/cura/resources/i18n

civic220, 2020-03-26:
> It would be my guess that your resources/i18n folder only contains .po files, and not the "compiled" .mo files. Those can be found here:
> https://github.com/Ultimaker/cura-binary-data/tree/master/cura/resources/i18n

thanks a lot！

closed 2020-03-26 by civic220