Ultimaker/Cura #7354: Improve brim breakability

Sophist-UK, 2020-03-26:
**Is your feature request related to a problem? Please describe.**
I am a relative newcomer to 3d printing, but love UC. I have successfully printed things with a brim to improve adhesion and stop warping, however I find that the brim doesn't break away cleanly.

To ensure the brim sticks, I use a Brim flow setting of 110% but I find that the brim does noty break away cleanly and needs trimming with a knife.

**Describe the solution you'd like**
I would like to be able to print the innermost thread of the brim at a different flow rate in order to improve the ability to break the brim away cleanly.

**Describe alternatives you've considered**
Insufficient experience to think of alternatives.

**Affected users and/or printers**
All.

mahtDFR, 2020-03-26:
Version 4.5 and up includes a feature by @smartavionics called 'brim distance' which works really well. Using a 0.4 nozzle and a brim distance value of around 0.5 creates a brim that basically lifts off the piece, while providing thermal benefits throughout the print.

As the feature as you describe it already exists, I'll close this off.

closed 2020-03-26 by mahtDFR

Sophist-UK, 2020-03-26:
I am using the latest version of Cura and using the brim distance (which is excellent). But this is not the same as the feature I was asking for.

mahtDFR, 2020-03-26:
Hmmm. Alternatively you could adjust the skirt/brim flow, but that doesn't quite fit your request either. Pull requests are welcome if you'd like to try.

Sophist-UK, 2020-03-26:
I will take a look at coding it at some point when I have time.

But the point of this FR is for **when** you use the Brim flow setting to have a thicker brim, but don't want this to make it difficult to break the brim off.

Ghostkeeper, 2020-03-30:
When you reduce the flow rate of a line, essentially what you're doing is reduce its line width but without reducing the spacing between the lines. Reducing the line width for the innermost brim line will indeed reduce adhesion to the model (as well as to the next brim line) since there is now a gap where there was previously material. I think reducing the flow of the innermost brim line by 30%-ish would have a very similar effect to having a brim gap of 0.1mm or so.

Sophist-UK, 2020-03-30:
You may want a thicker brim (at say 110%) for better adhesion, but not want to make it more difficult to break off the brim - so you might want to print the innermost brim line to 100%.

I don't think you would want to reduce it e.g. to 70% because that would probably void the whole purpose of the brim which is to help the main item to adhere to the bed.