Ultimaker/Cura #7355: After printing the brim or raft centered, the extruder then displaced and prints the model far from the centered raft. I have 2 Ultimakers 2+. How do I fix it?

SandraEMyers, 2020-03-26:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

SandraEMyers, 2020-03-26:
[image]
[image]
[image]
I have attached pictures of the Cura setting and the final 3D prints

SandraEMyers, 2020-03-28:
I had to stop the printer, suddenly started printing out of bounds towards the edge of the workplate!

Ghostkeeper, 2020-03-30:
The g-code you shared looks correct to me. It's not going outside of the bounds of the printer.
[image]

Have you considered that this could be a problem in your printer rather than in Cura's g-code output?

Ghostkeeper, 2020-03-30:
Reading over the bug report template you filled in, I think you intended to report a firmware bug, as you filled in the firmware version _3.3.0.

Perhaps you should look into layer shifts and the ways to prevent them: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/layer_shift.md

nallath, 2020-03-30:
I'm fairly sure that this isn't a slicer (or any software) issue. I think you should check if your pulleys (both on the rods and on the motor) are still tight. They are the most likely candidates for causing this.

SandraEMyers, 2020-04-07:
yes I checked every pulley, it seems ok. I'm still watching it just in case.

Best Regards,

SANDRA MYERS

Innovation Lab & Technology Educator

[cid:031609b4-6efe-4c01-95ef-1711c8b6ed85]

[image]<https://user-images.githubusercontent.com/2448634/77924828-79176200-72a4-11ea-83e9-50d413d4bb42.png>

SandraEMyers, 2020-04-07:
Thanks for your support, stay safe and healthy!

Best Regards,

SANDRA MYERS

Innovation Lab & Technology Educator

[cid:3c06cab7-64f1-4493-8258-b04800b6bee4]

[image]<https://user-images.githubusercontent.com/2448634/77924828-79176200-72a4-11ea-83e9-50d413d4bb42.png>

SandraEMyers, 2020-04-07:
I have read the article on layer shift, so far stop doing it since the last reset. I truly appreciated your support, I think it helped, Thanks!

Best Regards,

SANDRA MYERS

Innovation Lab & Technology Educator

[cid:336eb2c8-9673-462d-a641-7a11c7020e5f]

SandraEMyers, 2020-04-07:
Ghostkeeper also concurred. I did check the pulleys, they seemed ok. It hasn't done it since the last reset. Thanks so very much for the support!

Best Regards,

SANDRA MYERS

Innovation Lab & Technology Educator

[cid:05f53c50-11f9-4767-b5e1-5e2bee5d25f1]

Ghostkeeper, 2020-04-09:
Glad to hear you found the issue. We'll close this then since it's not a bug in the slicer but more a problem in the hardware and the settings.

closed 2020-04-09 by Ghostkeeper