Ultimaker/Cura #7358: Separate Material and Profile Settings

neomacedo, 2020-03-27:
**Is your feature request related to a problem? Please describe.**
Actually Material and Profile are completely "merged", what it means, if you have a good setup for a "product" to be printed and do you wanna print in different materials, you'll need to save different profiles, one for each material since if you just change the material into the dropdown menu the configuration set into the material won't be loaded (overriding the already set into the sidebar) if you choose to keep the customized settings

**Describe the solution you'd like**
As a 3D Printer guy, I wanna have a way to see the profile (named accordingly with each "product" specification) in a separate configuration than the material. Being able to use always the same profile (for the same product) but printing with different materials always when needed with a single change into the material dropdown menu

**Describe alternatives you've considered**
To have a checkbox enabling/disabling the material setup overriding

**Affected users and/or printers**
Everyone who print the same object using different materials

fieldOfView, 2020-03-27:
With the Material Settings plugin, you can put more settings in the Material pane of the preferences. You can get it from the Marketplace.

Ghostkeeper, 2020-03-30:
> if you just change the material into the dropdown menu the configuration set into the material won't be loaded (overriding the already set into the sidebar) if you choose to keep the customized settings

I don't understand what you mean with this, I think. My reproduction steps are these:

1. Duplicate Generic PLA, call it Generic BLA.
2. Change the printing temperature of BLA to 201 degrees (instead of 200).
3. Change to PLA. The printing temperature is 200.
4. Your step: "change the material in the dropdown menu" to BLA.
5. The configuration in the material is now loaded, as the printing temperature is 201 now.

The custom profiles are a separate configuration from the materials. As exemplified by these steps:

1. Change the infill density to 42.
2. Save the changes as a custom profile (e.g. called "higher density infill").
3. Change the material from PLA to ABS.
4. The custom profile "higher density infill" is still available and selected.

Could you clarify what you'd like us to change with these steps?

loneFunction, 2020-04-01:
@Ghostkeeper
Changing materials does change the appropriate temperature, unless you have changed the temperature in the profile at any point. The priority of profile vs material overrides is backwards.
I made this mistake before I knew about material configs and temperatures would not change when changing materials. In order to fix this issue I had to open the profile file manually and remove the temperature setting.
There should be an icon on the side of the temperature to revert to the material's setting.
I'm not sure if this is still the case in 4.5, but I have not changed the printing temperature from the sidebar since then and only setup new materials for different temperatures.

Ghostkeeper, 2020-04-02:
> The priority of profile vs material overrides is backwards.

I don't agree here. If the material would override the custom profile then the act of saving your current settings to a custom profile would make some settings suddenly ineffective since they are then overridden by the material.

> There should be an icon on the side of the temperature to revert to the material's setting.

With this I do agree though. Or at least, not specific to the material's setting (which is only one of the 13 profiles active on an extruder) but there should be a way to remove a setting from a custom profile. We have an item in our backlog to implement this, if I recall correctly. The "fx" button does this sorta but works in a different way, a limitation that doesn't make sense at all to the user.

loneFunction, 2020-04-02:
I don't know the specifics enough to really say about the priority of overrides. My experience with Cura is only with one single extruder printer. 
I do believe that temperature should be associated with the material and not the profile though. There is an old plugin which added a good feature to the profile that could modify the actual set temperature as an offset. So if you're profile is set faster and bigger lines you could increase the temp to compensate.
https://github.com/Pheneeny/CuraPlugin-PrintTemperatureOffset

With that kind of feature the material would always dictate the base level settings, with the profile modifying the output to accommodate the custom settings.

I can say as a new user it was very frustrating to setup materials due to the override behavior.

nallath, 2020-04-03:
You can find some more info about the profile system on https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings

hpandelo, 2020-04-18:
@nallath I got your point, the content it's really good for us, developers, but the Cura users isn't from our profile and the usability is completely compromised with this actual flow.
As a new user, I've got a lot of problems to "align" a good setup for the first material that I've bought (PLA EasyFill), after that, I've bought a different one (PETG), but sometimes I need to swap between the materials, and just changing the Material settings won't affect the full setup. This "nested tree" doens't makes sense to me as a user.

My idea is to have completely separate the Material Settings, the Printer Settings (Since I belive who are using multiple printer should have the same issue) and the "Product" settings.
Will be A LOT easier if just changing the Material dropdown, I being able to change my filament and get the same print result

nallath, 2020-04-20:
But how would that system work then? We've tried to come up with a system that works the way you describe it, but we always find some exceptions to it.

hpandelo, 2020-04-21:
A way to make it more simple (I won't call "a way to simplify" haha) is to remove the override rule from the print settings, since it appear to be completelly "magic" for the user. 
In my case here, I have a setup that I can't make it to respond to the material changes anymore, I haven't found a way to "fix" this.

**My Suggestions**
1. [Toolbar Improvements Suggestion](https://pasteboard.co/J4LDmu0.png)
2. [UI Improvements Suggestion](https://pasteboard.co/J4LD43C.png)

Ghostkeeper, 2020-04-22:
It says "image not found" when I click on those links :(

You can also paste an image directly in here, on Github.

Ghostkeeper, 2020-04-22:
> My idea is to have completely separate the Material Settings, the Printer Settings (Since I belive who are using multiple printer should have the same issue) and the "Product" settings.

To give an example of why this won't work, let's take printing temperature. Very obviously a "material" setting, most people would say. However when you're printing faster you're going to need to extrude more material per second out of your nozzle, and so it helps to make the material less viscous. Increasing the printing temperature is a good way to do that, so really printing temperature is a "product" setting, as it depends on whether the user wants to produce quickly or print beautifully.

Alternatively, take infill density. Most people would say that this is a "product" setting I guess, as it's clear that some parts are meant to be strong and some are meant to print fast or look good. But then reducing the infill density also causes the top skin to sag more. Some materials (especially high-temperature materials) sag way more than others, so you need a greater infill density for those to achieve a similar top surface. In fact, the infill in Cura's defaults is tuned way more towards preventing pillowing than to increase strength, at least with the Ultimaker profiles. I'd normally classify this as being more tuned to work with material properties and cooling than to work for strength.

loneFunction, 2020-04-22:
Yes, I would agree that baseline temperature would be a material setting, and there is also the need to adjust it with a product setting. That means one temperature box in the settings is not ideal with profiles fighting for control.
There is a plugin that handles this nicely which adds a printing temperature offset value to the main profile.
Switching materials changes the baseline temperature, and then the profile modifies it with an offset value. If printing with a larger nozzle, positive offset is added, but the base temperature never changes unless it does with the material.
This workflow makes a lot of sense in practice and has helped me keep material/profile settings without conflicts.
[image]

FYI this is not my plugin.
https://github.com/Pheneeny/CuraPlugin-PrintTemperatureOffset

hpandelo, 2020-04-23:
I'll re-add the images here

**My Suggestions**

1. **Toolbar Improvements Suggestion**

[image]

2. **UI Improvements Suggestion**

[image]

hpandelo, 2020-04-23:
@Ghostkeeper as @loneFunction suggested, we can have all overwrite needed as the offset field works (Controlling by percentage or temperature to increase/decrease). It will allow us to have a way to change filament type for example PLA to PETG, with the same print speed too. 

Into the software programming, we have the Immutability principle, the idea is to have the same into the UI, the material settings wouldn't be affected, allowing me to change the material regardless the print settings. 
In my case, I'll need to have the same print settings, with different materials, and it isn't possible today

Ghostkeeper, 2020-04-28:
So the way that this currently works is that the material settings adjust the [Default Printing Temperature](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/material/default_material_print_temperature.md) while the actual printing temperature used by the print is the normal [Printing Temperature](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/material/material_print_temperature.md). By default the Printing Temperature is equal to the Default Printing Temperature, but [some quality profiles override this to add an offset](https://github.com/Ultimaker/Cura/blob/68e8261c2124415814b89809fc8c914e2e4f3bde/resources/quality/ultimaker3/um3_aa0.4_PLA_Draft_Print.inst.cfg#L20). This way the user can still change the temperature in the material while the profile may still have an offset, or a multiplier, or whatever.
This implements basically the same feature as that offset plug-in.

Cura's profiles model is fundamentally different from your proposed immutability principle, because with that principle changing the material wouldn't have any effect any more. There is only one settings list, and different profiles may override parts of that settings list. If you'd want the custom profile to always have the same settings regardless of which material is underneath it, you could do that by overriding all settings with a fixed value (in theory). However then by definition the profiles underneath it don't have any effect any more so we might as well do away with the whole concept of having material or nozzle profiles. There is a way to reproduce a list of settings precisely though: File -> Save saves all of your settings, simply by saving which printer, nozzle and material is active underneath the custom profile, and so the settings list will reconstruct again as it was when you saved it.

closed 2021-07-26 by Ghostkeeper