Ultimaker/Cura #7359: Better control of dual extruders/Menu Suggestion

AbeFM, 2020-03-27:
**Is your feature request related to a problem? Please describe.**
Per https://community.ultimaker.com/topic/30393-copy-settings-between-extruders/
It oftentimes is hard when doing development to get your nozzles doing the same thing. Copy all changed values has a number of pitfalls. What started as an idea for "Copy ALL settings to extruder #n" got me thinking....

**Describe the solution you'd like**
[image]

**Describe alternatives you've considered**
Flipping back and forth. Many many times, on every screen over and over and over.

**Affected users and/or printers**
People with multiple extruders who make changes or get profiles from other people.

**Additional context**
A few small features I'd like to point out:
1) Some hidden options will need a space holder, as shown with tooltip.
2) Tough with an arbitrary number of extruders
3) Center linked options. It really helps them stand out.
4) I wanted to put material selection at the top... But filament changes can be complex.
5) I'd STILL like a button for copy all to other side
6) I think this would save a LOT of time flipping back and forth, even just to set settings directly.
7) Lastly, with existing system or this one, you could color code ones that match regardless of Linked status.

fieldOfView, 2020-03-27:
Nice mockup. There are some architectural problems that stand in the way of a quick test in a SidebarGUI-like way (such as that Cura is based on the notion of having one active extruder to get/set settings, instead of an easily accessible array of extruders to get/set settings). Another big problem is performance; Cura already gets very slow when a user has all/many settings visible, and almost doubling the number of QML elements to set up will not help that.

AbeFM, 2020-03-27:
The slow is the main reason I need it.

I could live with not-real-time. Set setting, don't apply till you hit
slice/save/etc. I'm sure you x could write 5 settings, change extruder,
write 4 settings from a list of 9 changes...

Maybe set it up to only show once section of variables at a time?

Similarly - what about a copy-all-to-extruder-N?

The issue is it's hard to tell which settings are different out of 50
trillion settings, you can only view so many at a time and it takes 10-15
seconds to switch per screen - even if you were a prefect robot it would
take several minutes to do.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Ghostkeeper, 2020-03-30:
We've had this idea before as well but it wasn't implemented for 3 reasons, all of which have been mentioned here already:

* It starts getting REALLY tight on space, especially with more than 2 extruders. Cura supports up to 16. Even with a modern screen you won't be able to fit those in, nor make it easy to use as you're basically looking at a big grid of settings then. Leave alone being able to see something of your 3D scene also. This could be solved by putting it in a 2D scrolling view, but that doesn't necessarily make it easier to use either, if you need to scroll back and forth instead of using tabs to go back and forth.
* We currently create a lot of interface elements once and then change out their contents depending on the "active extruder", i.e. the extruder tab you have selected. This includes all of the settings as well as the contents of the configuration tab where you select the material and nozzle. Instead of swapping out the material selection drop-down for another, we change the actively selected material that's shown in there. We'd have to refactor that. This is probably the architectural issue that fieldOfView is mentioning.
* We're afraid of the performance impact. While you might find it acceptable that the interface freezes for a few settings when changing a setting, many users would not like this.

hbh7, 2020-03-30:
How about a button to copy settings to another extruder? I just had to go through by hand switching back and forth comparing (with a few seconds frozen during each switch) and it was fairly annoying to do so. I like the idea of multiple extruders visible at once but I do see the potential issue with that. Perhaps make it configurable?

AbeFM, 2020-03-30:
> We've had this idea before as well but it wasn't implemented for 3 reasons, all of which have been mentioned here already:
Hopefully we can find a way to help the situation with a reasonable level of effort and impact. I hope it goes without saying but I value the discussion and desire it to lead to a implementable approach.

 
> * It starts getting REALLY tight on space, especially with more than 2 extruders. 
I mentioned, and restate: _2 at a time_ would already be a HUGE upgrade, and would work for MOST people. I have other machines with more nozzles, but didn't ask for a giant chart because that is unreasonable - a separate screen like a spreadsheet might work, but again, just because you CAN load 16 nozzles doesn't mean that making Dual work better would be pointless.
If I buy an Ultimaker today, how many nozzles will it have?

> * We're afraid of the performance impact. 
I'm not going to waste everyone's time with videos, but it takes 10-15 seconds oftentimes for me to switch to the other extruder. Not even to make a change, just to look. And I have to do it ~10 times to check all the settings, and rely on memory and eyesight to ensure I did it right.
I'm *already* irritated at the long, long pauses. If they weren't there this would be less of an issue.

The "**copy all settings to extruder #N**" button would help this problem with reasonable effort?

nallath, 2020-03-31:
The problem with writing software is generally not getting the _common_ cases to work. Those usually make sense (and thus, can usually be fit into whatever architecture is out there). It's usually the gritty fringe details (what if a user has 16 cores because technically, they can) that mess it up. As much as we would like to ignore those cases, we simply can't. We need to come up with some mitigation (so either select a set of them, provide some UI for that, whatever).

As for why you have these long delays and pauses, I have no idea what so ever. Are you able to run Cura from source? If so, you might be able to provide some profiling information so we can figure out what is going on with your system. 

Other than that: the copy all settings to extruder N should be pretty doable.

Ghostkeeper, 2020-03-31:
@hbh7 We did add that. It's there when you right-click on the settings list.

nallath, 2020-03-31:
It's only the changed settings though. I'm not sure if that is what he wants.

Ghostkeeper, 2020-03-31:
Copying all settings to extruder N is the same as copying all settings to all extruders in the case of a 2-extruder printer, going by AbeFMs argument.

Ghostkeeper, 2020-03-31:
Indeed, you'd also need to change the profile on that other extruder.

AbeFM, 2020-03-31:
Jamie: yes, copying to Extruder N (or to all) would work. It's my "runner
up" fix, and I'd be ok with chasing this thread to reflect that that's
where this is headed.
It could be extended with something like locked/protected settings, but
even just a copy would be a big step in usability.

Copying to ALL extruders seems problematic, if I have 16 cores odds are
good one of them is something different (supports/temps/retract/etc) than
the rest.

One last suggestion - could you load settings to one particular extruder
instead of to the .... Whatever you call it. The build environment?
...Instead of overwriting all extruders, let me load settings for just one
:-)

 I assume you do something like that when an UM loads a different "core"?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Ghostkeeper, 2020-04-01:
You mean, like a custom profile but only for one of the extruders?

Ultimaker's Print Cores are to Cura just the same as what other printers call their nozzle. Cura has a profile for it that you can swap out, and the availability of materials and quality profiles can depend on the selected nozzle. Physically Ultimaker's print cores also have a few other things than just the nozzle (like a temperature sensor and heater element), but to Cura it's all the same, just a different term we have to display.

AbeFM, 2020-04-01:
Exactly, I'd like to load "my dumb left nozzle filename" to one nozzle
without overwriting non-shared settings from other extruders.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Ghostkeeper, 2020-04-02:
Then you'd also need to have the profile selector be per-extruder though. And all of the global settings would need to match with whatever profile is selected in the other extruders. Some profiles are just for one of your extruders and some are for all of the extruders. Sometimes the other extruders switch along with the profile change and sometimes they don't. I think this would become confusing very quickly to an average user.

AbeFM, 2020-04-02:
This seems to be focusing on how it couldn't work instead of what would
help.

So far, I've seen no serious objections to a button/option that copies
settings from one extruder to another.

Is that something we could seriously consider?

>

Block137, 2020-04-04:
> Then you'd also need to have the profile selector be per-extruder though. And all of the global settings would need to match with whatever profile is selected in the other extruders.

I like this profile-per-extruder idea. As for global settings, could use the profile change warning dialogue which ask whether to "keep current global settings" or "discard/overwrite global settings with values from new profile". Default to using settings from T0.

Ghostkeeper, 2020-04-06:
I think I probably sound negative pretty often. That's because it's part of my job to judge whether it is feasible to implement this, taking into consideration all of the current requirements, the user experience, not damaging current users' configurations and the development effort vs. reward consideration. I think the idea of being able to load different profiles per extruder fails on 3 of these fronts: Current requirements, user experience and development effort vs. reward.

I'm saying current requirements because the sliders in Recommended mode currently change the profile for all extruders and there are no tabs there to change this per extruder. It also wouldn't work per extruder since there are also global settings such as layer height that need to be the same for all extruders, so you wouldn't be able to save those in a profile any more.

I'm saying user experience because while the regulars here on Github are decently experienced with computers, many of our users have no experience with software. They wouldn't understand that if they want to use a profile, it's not immediately applied to the entire print but only to parts of the print. Oh it happens only to the ABS parts, that's weird. Must be something wrong with this printer. They won't realise that they would also need to go into that other tab and change the profile there.

I'm saying development effort vs. reward since Cura's profile system is about the most complicated profile system you'll see in consumer-oriented software. About a third of our code base is dedicated solely to this. Much of that complexity stems from multi-extrusion support. Changing something in this regard is dangerous and expensive. That's not a consideration for your feature request itself (it's still a valid feature request even if it takes a long time to make) but it is a consideration for our priorities.

Ghostkeeper, 2020-04-06:
The idea of being able to copy settings from one extruder to a certain other extruder (as opposed to from one extruder to ALL other extruders) has more merit. It doesn't interfere with any current requirements since it's a separate function that applies a change to the settings when you click on it. It also doesn't affect the user experience much since we can hide it in a context menu.

The effort vs. reward is maybe a consideration. It's probably not a whole lot of work since we can reuse the code that we currently have to copy the extruder's settings to all other extruders. Just loop it less often. But then again, it's only effective for printers with 3 or more extruders, which are currently very rare in the field.

Block137, 2020-04-06:
I understand. Implementing per-extruder will completely break compatibility of current profile system (for multi extruder at least). I can imagine a big mess when converting old profiles (if importable at all).

Ghostkeeper, 2020-04-06:
Actually, converting old profiles is probably not a huge problem. In the configuration folders they are stored with a different file for each extruder. There is a problem of what you'd do with the global section of those profiles though. That's the same problem as what I wrote down under "current requirements" I guess.

DkOuvE, 2020-04-15:
I also like the idea of a per extruder settings. Then you can have global settings for the printer, and then settings for the filament that is loaded into the printer. Ideally, the per extruder settings can modify all settings and not just material related ones, like what's seen in Mattercontrol 2.0, as for example TPU would print at a lower speed than PLA and may need different acceleration settings to get a good print.

AbeFM, 2020-04-15:
> The effort vs. reward is maybe a consideration. It's probably not a whole
> lot of work since we can reuse the code that we currently have to copy the
> extruder's settings to all other extruders. Just [...]

The problem isn't one extruder versus all, but the current function copies
all CHANGED values.

My complaint is that the extruders may have differences I cannot find.

Another option would be to highlight differences between them - and
unmatched settings highlighted in a color. Even if only when you activate
this feature, it would really help.

But to follow in your thought - copy ALL settings, not copy some settings.

>

Ghostkeeper, 2020-04-22:
Yeah you'd also need to change the material and nozzle profile to be the same in the other extruder.

ok4global, 2020-04-28:
> We've had this idea before as well but it wasn't implemented for 3 reasons, all of which have been mentioned here already:
> 
> * It starts getting REALLY tight on space, especially with more than 2 extruders. Cura supports up to 16. Even with a modern screen you won't be able to fit those in, nor make it easy to use as you're basically looking at a big grid of settings then. Leave alone being able to see something of your 3D scene also. This could be solved by putting it in a 2D scrolling view, but that doesn't necessarily make it easier to use either, if you need to scroll back and forth instead of using tabs to go back and forth.
> * We currently create a lot of interface elements once and then change out their contents depending on the "active extruder", i.e. the extruder tab you have selected. This includes all of the settings as well as the contents of the configuration tab where you select the material and nozzle. Instead of swapping out the material selection drop-down for another, we change the actively selected material that's shown in there. We'd have to refactor that. This is probably the architectural issue that fieldOfView is mentioning.
> * We're afraid of the performance impact. While you might find it acceptable that the interface freezes for a few seconds (having been a user since the interface was much slower) when changing a setting, many users would not like this.

Ghostkeeper, thanks for the explanation but the current interface isn’t conducive to using the machine to its fullest. So much time it lost to checking and double checking settings, and even more time is lost to print mistakes because a single setting if off. It doesn’t seem necessary to see the print bed while looking at the settings. Nothing changes until it gets resliced. Instead of trying to fit everything one one screen, why not have the settings be on a dedicated screen. It would be much easier to design an intuitive interface. Please make it easy for users to use the amazing machines to their maximum ability!  Please make it intuitive so the wonderful complexity of the software and machine  becomes accessible to everyone.

AbeFM, 2020-04-28:
Yeah, that's what I was trying to say. Even if I have to open a
separate screen/environment, it's better than hitting print and then not
knowing what's different. It's why I suggested a highlight, which you could
do without any other changes.

Another idea, to address the slow speeds, is to only allow one section
(e.g. "travel" or "speed") to be displayed at a time. I get it, anything
you change will impact something else, but if the reason you're not doing
it is it's hard to do, then let's work on that.

If it's because you don't find the suggestion to be of general benefit, we
can talk about that. I, as a member of the user community, would like to
work with you guys until we get to a point where there's a feasible path
forward to remove the issue of having mismatched settings between extruders.

Ghostkeeper, 2020-04-29:
To clarify, we do understand the benefit of such a change. The hurdles are kind of hard to overcome though. We need to ensure that it's actually better. If you can only show one category of settings at a time, the setting search functionality would be pretty much broken. Many users use this feature way more than the actual categories or the setting visibility.

We're already getting lots of complaints about performance of loading the list of settings and that the available screen size is too small. There are changes that we can make to fix this to be sure, but all of them require very significant redesigns and rework, which puts this on lower priority for us than, say, improving the quality of line fitting and gap filling algorithms in CuraEngine. It's not a low hanging fruit, you see. The hurdles that I wrote down are significant, all three of them.

ok4global, 2020-04-29:
I confess I am not understanding what is hard to implement about AbeFM's two column design.  

_- It starts getting REALLY tight on space, especially with more than 2 extruders. Cura supports up to 16. Even with a modern screen you won't be able to fit those in, nor make it easy to use as you're basically looking at a big grid of settings then. Leave alone being able to see something of your 3D scene also. This could be solved by putting it in a 2D scrolling view, but that doesn't necessarily make it easier to use either, if you need to scroll back and forth instead of using tabs to go back and forth._

I didn't know that Cura supports 16 extruders (wow!), but what percentage of your users use more than 2?  And, why is it valuable to see the 3D scene while changing settings?  The scene does not change until it gets resliced.  If someone needs to look at the scene to learn something, they can hide the settings panel, just like they can now.  Seeing all the settings at once is much more important than keeping the 3d scene in view simultaneously.

Also, it took me a long time to understand your comment about a 2D scrolling view – I gather you mean scrolling horizontally to allow for multiple columns?  Yes, that would make it **much* easier and solve the problem.  The issue is being able to compare setting between extruders.  It empowers experimentation and learning, and also helps prevent mistakes.  It is so easy to change one extruder and forget to change the other, or to loose track and think I am change a setting for extruder 1 when I am actually in extruder two tab, or not to realize that a particular setting doesn't have the subtle icon indicating that it is global.  It is asking too much of the user to remember everything. 

_- We currently create a lot of interface elements once and then change out their contents depending on the "active extruder", i.e. the extruder tab you have selected. This includes all of the settings as well as the contents of the configuration tab where you select the material and nozzle. Instead of swapping out the material selection drop-down for another, we change the actively selected material that's shown in there. We'd have to refactor that. This is probably the architectural issue that fieldOfView is mentioning._

Sorry, you lost me.  

_- We're afraid of the performance impact. While you might find it acceptable that the interface freezes for a few seconds (having been a user since the interface was much slower) when changing a setting, many users would not like this._

Again, not sure how multiple columns affects performance.  I thought they were just a list of numbers.  When I change a setting, does something happen in the background before I resclice??  It doesn't seem like it.  If I change infill from 5% to 90%, I don't notice any slow down in the settings menu, although reslicing takes a long time.

Also, with multiple columns, search would still work to find a setting.

DkOuvE, 2020-04-29:
The problem is that there's probably a lot of people that have MMU2s or 3 in 1 outs. And these days toolchangers are seeing an uptake. I'm currently trying to design an 8 filament device similar to the MMU2 and there's already a massive performance hit just telling Cura that there's 8 extruders on a printer. Seems that it stems from the amount of textbox elements shown in the settings list, and I have a lot of them visible, which means even displaying 2 values is probably going to bring things to a halt.

Could someone inform me of the issue of my previous suggestion which was the profiles applying to individual extruders and having a global and "per-extruder-setting" tab instead of the extruder tabs, where the per extruder tab shows all the settings that aren't the same on all extruders and the ability to add additional settings like the per model settings feature? Another similar idea could be sub-profiles for the extruders and the main profile for all the mandatory shared settings, resulting in 3 tabs. Then you can have invariant settings on the main profile and a PLA / PETG / ABS sub profile for all the specific changes for that material, and then per extruder settings for fine changes.

Ghostkeeper, 2020-04-30:
> And, why is it valuable to see the 3D scene while changing settings?

You should hear the amount of criticism we got for moving the settings list to a collapsible panel because it was 20 pixels further to the left. People use small screens, especially where they have computers stationed specifically next to their printers. Most printers indeed have 1 or 2 extruders, but to display 2 could already be a problem.

Like I said though, horizontally scrolling could solve that but could also defeat the idea, in part. It doesn't solve the rest of the issues mentioned here though, since you can then still only see a single extruder at a time and won't be able to see the global settings at the same time sometimes.

> > We currently create a lot of interface elements once and then change out their contents depending on the "active extruder", i.e. the extruder tab you have selected.
> Sorry, you lost me.

It's an architectural issue. We don't re-draw the settings list on the screen when you switch extruders. We only have one presenter for the current extruder so every time something changes we need to only update one presenter with whatever the new setting values are.

> Again, not sure how multiple columns affects performance. I thought they were just a list of numbers. When I change a setting, does something happen in the background before I resclice??

Yes, it needs to run [this algorithm](https://github.com/Ultimaker/Cura/blob/master/docs/profiles/getting_a_setting_value.md) (minus the CuraEngine paragraph) for every setting that depends on the setting that you changed, for all visible settings.

Ghostkeeper, 2020-04-30:
> Could someone inform me of the issue of my previous suggestion which was the profiles applying to individual extruders

This is how it worked in Cura 2.2 but we found that it was too complex for a user to understand. There would be many cases where the user switched to a certain profile but forgot to change it for a different extruder.

> and "per-extruder-setting" tab instead of the extruder tabs, where the per extruder tab shows all the settings that aren't the same on all extruders and the ability to add additional settings like the per model settings feature?

That would make sense for an expert that is optimising his multi-extrusion prints and wants to keep some differences between extruders due to model geometry (where just copying all settings to the other extruder doesn't cut it). But it wouldn't make sense for cases where you have a different material or a different nozzle in the other extruder because then many settings will be different.

> Another similar idea could be sub-profiles for the extruders and the main profile for all the mandatory shared settings, resulting in 3 tabs.

Internally we store the custom profiles like this. Except that for global settings we display them in the same list with a chain link icon, because that's easier to oversee and understand than having global settings in a separate tab.

DkOuvE, 2020-04-30:
> But it wouldn't make sense for cases where you have a different material or a different nozzle in the other extruder because then many settings will be different.

Yeah I can see how if it shows every different setting, this idea only really works with the main profile + filament modifier profile + per extruder idea, where it'll be something like

- Extruder 1 V
   - Nozzle temperature [225]
   - _+ Add setting_
- Extruder 2 <

Not sure if it'll look cluttered on smaller screens, or the performance hit from more icons, but it might be a step in the right direction if it's possible to toggle if values are shared among extruders, and a broken link is shown instead so that you can hover over the icon and see all the other values at a glance. It won't fix dealing with PLA in both extruders or PLA in 1 extruder and PETG in another with the same profile, but it helps with keeping settings the same for the same material in all extruders.

AbeFM, 2020-04-30:
> Could someone inform me of the issue of my previous suggestion which was the profiles applying to individual extruders and having a global and "per-extruder-setting" tab instead of the extruder tabs,

> You should hear the amount of criticism we got for moving the settings list to a collapsible panel because it was 20 pixels further to the left.

Oh yeah, that bothered me too, though on my 4k just living with it is reasonable. Can you believe I bit my tongue on something? :-P

> Like I said though, horizontally scrolling could solve that but could also defeat the idea, in part. It doesn't solve the rest of the issues mentioned here though, since you can then still only see a single extruder at a time and won't be able to see the global settings at the same time sometimes.

I like this though I'm *starting* to get how it would be hard.
I really like the idea of a separate screen like a spreadsheet with each extruder on a collum. This, DkOuvE's suggestion above, all of these could work on a screen you'd pull up just to check it. If it takes me 2 minutes to slice and then I have to GUESS what's wrong, I could wait 1 minute for a comparison to load up so I can SEE what's wrong - without impacting the traditional user experience. Admittedly it's a chore, but it addresses the performance/screen space trade off with features.

> Internally we store the custom profiles like this. Except that for global settings we display them in the same list with a chain link icon, because that's easier to oversee and understand than having global settings in a separate tab.

Ah, yes, you're almost doing what I suggested then - using colors (or some mark) to distinguish linked.

_Take it a step further - black chain for linked, green check for "all extruders using the same value, but it's not linked", blue hollow dot/square for "different values on different extruders"_

I believe that would address the issue without significant impact. It takes up no extra space, and while it does require you to know what all the values are at run time, it doesn't require you to swap out listsboxes or anything. Even if you drew in the screen as is with those missing, THEN did a second pass to fill them in, people could wait for it.

DkOuvE, 2020-04-30:
> Take it a step further - black chain for linked, green check for "all extruders using the same value, but it's not linked", blue hollow dot/square for "different values on different extruders"

Having coloured icons would mess with the entire aesthetic of the list. I'd have 3 different looking icons.

> This is how it worked in Cura 2.2 but we found that it was too complex for a user to understand. There would be many cases where the user switched to a certain profile but forgot to change it for a different extruder.

What about the MatterControl method? Instead of the material tab (that doesn't really store a lot of settings for actually printing the material) being somewhat separate, having the materials as a subprofile of the main profile that includes all the changes that you'd make for different filaments

Being able to add custom setting calculations would also be nice, but that's another topic.

AbeFM, 2020-04-30:
Take it a step further - black chain for linked, green check for "all
>> extruders using the same value, but it's not linked", blue hollow
>> dot/square for "different values on different extruders"
>
> Having coloured icons would mess with the entire aesthetic of the list.
> I'd have 3 different looking icons.

Sure, at this point I'm trying to find out what's technically feasible for
the team. Personally I would welcome the colors, or better still, highlight
the cells with a pastel half-tone kinda thing... But really I want to know
if the idea of displaying "linked/uninked" could be changed to
"unmatched/matched/linked", somehow. Something the team can implement.

Awaiting comments from the devs.

Ghostkeeper, 2020-05-01:
> Take it a step further - black chain for linked, green check for "all extruders using the same value, but it's not linked", blue hollow dot/square for "different values on different extruders"

While the idea of a chain icon with a different colour itself may not be very clear to a user, having some sort of symbolism to indicate this has some merit to solve the problem of the layout and space claim. I like that idea.

It would still require us to know which settings are the same as settings on other extruder stacks. Without trying that out it's pretty difficult to guess how hard that would be to make. It will cost quite a lot of performance though when loading the settings list and when changing settings. I don't think there is really a way around it.

ok4global, 2020-05-01:
For what its worth, I do not find the current chain icon very useful, and multiple colors would not enhance its usefulness. Yes, better than nothing, but it is very easy to overlook or forget about or ignore.

I just read an article in the New York Times, which seems insightful to this specific question, and globally to all software design.  It makes the case that the best design, best software, best tech doesn't win.  Rather easy wins.  In case it is interesting for you, here it is. 

[No, the Best Doesn’t Win - The New York Times.pdf](https://github.com/Ultimaker/Cura/files/4564055/No.the.Best.Doesn.t.Win.-.The.New.York.Times.pdf)

DkOuvE, 2020-05-01:
I'm trying out the Material Settings Plugin that - since it can changes all non global settings - essentially acts as a per extruder profile:

[image]
[image]
_Example for the Tevo Flash_

If you've changed a setting and saved it, there's no way in the UI to revert so that the plugin can take control (you have to remove it from the cfg file), but it is a massive step forward in material management and keeping the same settings between extruders for the same material.

Maybe a similar plugin can be designed to choose what settings are linked/unlinked so that there's a working solution out on the Marketplace for the users that need it and business as usual for the users that don't.

AbeFM, 2020-05-01:
Could you make them all linked by default, and you'd click the chain to
specifically unlock them?

Then we'd still have to flip back and forth to check, but it would be
easier to at least know which ones wouldn't be different.
I'd display an icon for *unlinked* variables only, and just leave the rest
blank. When you skim, you'll see it. Doesn't depend on other settings,
doesn't depend on number of extruders, would solve a lot of issues and
perhaps look cleaner than what we have now.

Ideally, a display set that shows only unlinked variables (even if only one
extruder) would round out the 'solution'.

Thoughts?

(Also, I like highlighting the field over coloring the link icon. But it's
still technically limited at this point, not suffering for lack of vision)

fieldOfView, 2020-05-02:
> Thoughts?

That could be nice for "dual color" printing, but horrible for dual material printing such as printing with soluble  supports. Given that that has been the biggest usecase for dual extrusion for Ultimaker 3 and up, I don't think it will go down well.

I also don't think adding link icons on all settings will work well. You won't see the forest for the trees. Finally I don't think it is a good idea to fundamentally change how the settings behave with just a small icon showing the difference.

Even with all the implementation issues mentioned, I like the original idea mentioned in the issue better, and may do some experimenting in the coming weeks to implement that in a plugin to see if it can be done in a performant way.

DkOuvE, 2020-05-02:
> Thoughts?

I'm for matched being no icon, icon for linked and icon for unmatched where you can hover and see different values.

ok4global, 2020-05-02:
I suggest going back to the beginning of this thread and looking at @AbeFM ‘s mockup of two columns. It is so simple and intuitive. Everything is understandable at a glance.

DkOuvE, 2020-05-02:
How does it work if you've got 3 or 5 extruders? Is it like you see extruder 1 against one of the other extruders?

ok4global, 2020-05-02:
> How does it work if you've got 3 or 5 extruders? Is it like you see extruder 1 against one of the other extruders? Perhaps like diffchecker for settings?

Most people don’t have 3 or 5 extruders, but for those that do, 3 or 5 columns.

DkOuvE, 2020-05-02:
> Most people don’t have 3 or 5 extruders, but for those that do, 3 or 5 columns.

Will they always be visible, or a button to "View All"?

AbeFM, 2020-05-23:
> I suggest going back to the beginning of this thread and looking at @AbeFM ‘s mockup of two columns. It is so simple and intuitive. Everything is understandable at a glance.

It was mentioned that the data itself isn't handy, somehow, which I'll admit I don't understand. I would really like to see it, and yes, if you have MORE than 2 extruders - say you dislike UItimakers for some reason - then you could pick which two to show or have a spreadsheet like screen... But those are major changes.

Being able to click any setting and force it to link would solve the issue.

What are the dev team thoughts? I again had an issue where extruder 1 printed correctly and extruder 2 had infill speed set to 100 mm/s (from the PLA profile) which caused big issues since I was running TPU....My 'copy all changed values' happened at the wrong time, it's vague and a little dangerous.

AbeFM, 2020-05-23:
FOV:
> That could be nice for "dual color" printing, but horrible for dual material printing such as printing with soluble supports. Given that that has been the biggest usecase for dual extrusion for Ultimaker 3 and up, I don't think it will go down well.

I don't understand what you're saying. I use dissimilar material much of the time and I absolutely see the use case. I'm not saying "only give us one setting", I'm saying make it possible to EASILY verify that they are ALL tied together except overrides.

> I also don't think adding link icons on all settings will work well. You won't see the forest for the trees.

Making them different color - or highlight the fields as I've suggested repeatably - will make it easy to see. What ISN'T easy is clicking the other extruder, waiting seconds for it to load in, trying to remember what every number on 5 pages of settings were and spotting the difference.

> Finally I don't think it is a good idea to fundamentally change how the settings behave with just a small icon showing the difference.

This already happens, only with a small subset of settings.

> Even with all the implementation issues mentioned, I like the original idea mentioned in the issue better, and may do some experimenting in the coming weeks to implement that in a plugin to see if it can be done in a performant way.

I'll admit I like my own idea. Looking forward to seeing what you have in mind and anxious to try stuff out.

Ghostkeeper, 2020-05-26:
> What are the dev team thoughts?

Mine are still unchanged. It's hard to make things fit on the screen, and it's quite a refactor to change our models. I've said these things before:
* A scroll bar is possible to make it fit on the screen, but kind of defeats the purpose if only one column can be seen at a time then. Might as well click on the tab instead of dragging the scroll bar.
* We still intend to support >2 extruders.
* Colours are possible as an alternative way to show when settings are different between extruders, but also quite ugly and jarring, like other people here have said.
* Icons are also possible, and less invasive than colours, but still vague to understand.
* In either case, with two different material types the setting lists are so different that the settings list will always be coloured like a Christmas tree.

There is also the issue that there is little incentive to make a big interface change like this for us now. I'd rather work on changes that improve print quality now, and our project managers prefer to see changes w.r.t. cloud connectivity because buzzwords.

> It was mentioned that the data itself isn't handy, somehow, which I'll admit I don't understand.

To help you understand the difficulty of this refactor, maybe this analogy helps: It's like you have a TV screen in an IKEA TV cabinet, and you want to be able to play with two different consoles. Currently we have it set up with a receiver so you need to switch with a remote which console input is shown on the TV. You're asking us to be able to play with both consoles at the same time. That means that we'd have to buy a second TV and somehow make both TVs fit in the same cabinet which is surely going to take a lot of carpentry, and that we have to switch out the receiver system with something else. And then it would also take a lot of time to turn both TVs on every time you want to play a game with your friend.

This analogy isn't perfect, because "switching out the receiver" is what would be the most work for the developers, but for the rest it's pretty accurate.

AbeFM, 2020-05-26:
Thanks for laying it out. Most of it totally makes sense, it's why I'm
asking - again, thanks.

The bit I don't agree with:
Colors are too obvious, and icons aren't obvious enough.

If that's REALLY what's going on, then we can - as a group - try to come up
with something that meets these art-department criteria. It seems there's
room there to make something that works. Light and dark grey shading
would work, for instance.

But as it is, it takes 5-10 minutes to check everything. If I send you a
profile, can you tell me how many settings don't match? How would you even
accomplish this?
Currently I find issues like this by printing, looking at the failed print,
and using reason and a developed understanding of each and every setting to
come up with a list of things I should check. I don't find it user friendly
AT ALL, and the other people posting here agree.

Improving print quality is obviously important, I can't set your
priorities. But this is an area - especially as I get into sharing
profiles, where the program makes things quite a bit harder than they need
to be. Either allowing the highlighting to turn on and off, choosing less
offensive colors, or yes, using B&W icons (I can look at a collum of icons
and spot the different one a LOT easier than memorizing 300 settings and
then clicking something, waiting 5-30 seconds for the screen to update, and
checking 300 more to look for differences.)

I wouldn't beat this up so much if it wasn't a constant problem. I don't
feel "it would be jarring if we solved the problem" is a great solution.

Again, thanks for all you do, no offense, etc etc etc ad naseum. :-)

Ghostkeeper, 2020-05-28:
> The bit I don't agree with:
Colors are too obvious, and icons aren't obvious enough.

Almost. I think both are not obvious enough as to what they mean when you see them. And I think that especially the colours would be very jarring and ugly in the interface. The icons maybe too if there are a lot of differences, like with different nozzle sizes and/or different materials.

It's a LOT of stuff that 99.9% of the users won't need to see.

But yes I am also hoping that we can find a solution together here. Whether we get the time to work on that solution or not is a different matter then, which will depend on the solution and on how big the problem is for Ultimaker's target audience. But at least it's a prerequisite.

ok4global, 2020-05-28:
I think the problem is much simpler than everyone realizes.  Right now, there is no easy way to look at the settings for both (or many) extruders simultaneously. If changing the software per @AbeFM's excellent initial mockup is complicated, then just it would be almost as useful if I could view a report that looked like the mockup.  

It would give me the information in a format which would empower me to quickly scan down all the settings and see which are global, which are unique, which are the same, which are different.  It would let me catch mistakes and develop much better experiments.  

Then I could go back to software and change what I need to.

It might help if I describe the problems differently: 

1. Mistakes in settings cost lots of time and money and failures.  I want a tool which allows me to easily catch mistakes.

2. Cura is powerful, but accessing the power is hard.  It is just a nightmare to use the current setup to experiment with the settings.  A tool which allowed comparing all the settings would empower users to experiment and use more of what Cura can do.   

The report could be simply another screen in Cura.  It would be ideal if it could be saved as a spreadsheet, but honestly it could also be a PDF.  

It would also give me a way to save a record of settings so in 6 months, when I have to print a similar 5 day project, I could go back and see what I did before.

AbeFM, 2020-05-28:
[image]

> Almost. I think both are not obvious enough as to what they mean when you see them. And I think that especially the colours would be very jarring and ugly in the interface. 

I really have trouble seeing your point here. There are ALREADY icons there that show up, on everything, with hard to spot differences but they work. 

Currently I see F(x), LINKED, Reset, and blank. We're talking about adding one more symboly here. If the argument is B&W symbols are unusable, you've got 3 to get rid of. If the argument is that one more will make it unusable I think that's a bit extreme. Especially since it'll ONLY show up for multi material users.

>The icons maybe too if there are a lot of differences, like with different nozzle sizes and/or different materials.
You can make this as subtle or overt as you'd like. ANYTHING is better than not having the information - as ok4g mentioned: Wasting material and days is a LOT more likely to send users elsewhere than a lack of cloud computing buzzwords. What I don't want is to tell a customer their part is late because the guys who wrote the software I use thought the interface would be jarring.

I think you'll have to add one differentiator to capture this. So far, I've seen no comments that say if we can't have it in real time we won't take it - so loading times don't matter, screen clutter doesn't matter. I'm not kidding when I ask: How do you CURRENTLY compare a profile I send you with a bunch of different extruders and a bunch of different settings - how do you tell if there's a problematic mismatch? I feel that the current approach doesn't make this process reasonable.

Consider this like a DRC in a schematic - I don't want to print with potential unseen errors. The current set up hides problems. Again, this isn't meant to be disrespectful - it's just the reality and thankfully it can be solved.
> 
> It's a LOT of stuff that 99.9% of the users won't need to see.

Again, not a great argument: Strongly wanting to support people with 10 extruders but saying people with 2 are out of luck because it is too rare? Considering your own machines are mostly dual nozzle that falls a little flat.

>...which will depend on the solution and on how big the problem is for Ultimaker's target audience. But at least it's a prerequisite.

The target audience is either (as we're saying) people using two, or more, materials and want to get a good print or (as it's being suggested) people who would be so offended by colors or icons that they'd use a different product if they were subjected to "matched" icon while printing with two different materials.

Sorry if it is ranty. Hopefully the points carried through. :-)

AbeFM, 2020-05-28:
> 1. Mistakes in settings cost lots of time and money and failures.  I want a tool which allows me to easily catch mistakes.
> 2. ... A tool which allowed comparing all the settings would empower users to experiment and use more of what Cura can do.
> 
> The report could be simply another screen in Cura. It would be ideal if it could be saved as a spreadsheet, but honestly it could also be a PDF.
> 
> It would also give me a way to save a record of settings so in 6 months, when I have to print a similar 5 day project, I could go back and see what I did before.

I really agree with all of his points here. I don't know who values money over time versus the other way around, but printing to catch mistakes is just NOT what any user wants.

And looking at two different profiles, be it nozzle to nozzle, user to user or old versus new.... I OFTEN have this issue with CURA.

Perhaps this could be done as a plug-in? As a check you RUN, not a passively updated indicator. I'd prefer the opposite, but that would meet the need.

DkOuvE, 2020-05-29:
Yeah I'm also thinking of a plugin, just so that we don't have to worry about the everyday consumer or Ultimaker's buzzwords and only the people that want the feature has it and has a better chance of knowing how to use it.

Ghostkeeper, 2020-06-02:
I hope you're not trying to argue against me specifically. I'm just trying to think along. To think ahead of the problems we're going to encounter and to advocate for the devil. That's part of my job after all.

> I'm not kidding when I ask: How do you CURRENTLY compare a profile I send you with a bunch of different extruders and a bunch of different settings - how do you tell if there's a problematic mismatch?

I don't. If I need to make a dual colour print where two extruders are the same, I'll make the changes in one extruder and copy those changes to the other extruder. If then some settings need to be *slightly* different for the other extruder, I'll copy them first and then make those slight changes. If I'd really need to compare them I would just open Cura twice and see both tabs at the same time, which works fine for me since I have a wide screen and don't mind waiting for Cura to start twice. But I normally rely on the built-in profiles and adjust 5-10 settings at most.

I see your point with your example drawings, but it's still going to be a mystery why some settings are purple or yellow unless people have read the change log from when this feature got introduced. We have the same problem now with the orange warning colour to some extent. Some people are confused as to what that means. And similarly what the Fx icon means. With the orange there is at least the cultural intuition due to traffic lights.

> Strongly wanting to support people with 10 extruders but saying people with 2 are out of luck because it is too rare? Considering your own machines are mostly dual nozzle that falls a little flat.

I think you misunderstood me. I didn't say that 2 extruders is rare, only that the use case of needing an overview of which settings are different is rare. We fully expect the market to grow towards >2 extruders in the future, although this isn't the case yet.

> Perhaps this could be done as a plug-in?

As a check you run would make the implementation considerably easier. You can look to the [God Mode](https://github.com/sedwards2009/cura-god-mode-plugin) plug-in and implement something similar.

AbeFM, 2020-06-08:
I'm done posting on here if EVERY SINGLE THREAD involves having to defend
my intentions.
I have better things to do with my life than to discover room for
improvement in CURA, spend all this time writing it up and then carefully
responding and gathering community support just to be accused of it being a
two and a half month troll.

ok4global, 2020-06-08:
I agree with @AbeFM ’s frustration. Your job might be to think of all the problems, but if Cura  hopes to encourage feedback, this forum should encourage all input. When I worked in city government, we knew for every written feedback we received, there were 10s of people with similar Thoughts who didn’t write in. 

To succeed, Forums like this should encourage and support all users who write in and make us feel like our contributions are valuable. Then, internally you can figure out the problems and solutions and decide if the solution is worth the effort. Otherwise, users lose interest and feel unappreciated.

DkOuvE, 2020-06-12:
If I knew how to do it I would have already begun coding up a popup window plugin that is like a diffchecker for extruders, where all the visible settings are shown in one large list with either no icon, linked or matched in the middle of each of the settings. Not sure if it's as easy as just adding the value that you want linked from the extruder cfg to the global cfg but if it's possible to do in a plugin, I'd also add the ability to link values.

Ghostkeeper, 2020-06-15:
Oh, no, I'm not offended or anything like that. Don't worry. Our goals are aligned here. I just don't have any answers to the problems I've posed, and as long as I'm responding, other developers will glance over this topic thinking that I'm taking care of it.

ok4global, 2023-11-12:
Hello,  Please, please, please review this issue.  Cura is so powerful and offers so many settings, but it is super hard take advantage of them and 3 years later, the user interface for extruder settings is still very hard to use and prone to errors. 

I'll rephrase it as a question: how can Cura user interface be improved so that users can successfully access the amazing controls in a way that maximizes the chances of success, makes tweaking settings easy, and reduces the likelihood of error?

Right now, Cura's user interface for settings is super complicated.  Other than the tab at the top of the screen, there is no differentiation between settings for Extruder 1 and 2, so it is super easy to lose track and accidentally change the wrong extruder.  And there is no way to see the settings for the 2 extruders side by side.  

The feature request is just a UI improvement, but I think it would really help users a lot and make Cura much easier to use. 

I reattached @AbeFM's mock up from 2020, because it is still a simple elegant solution.  Most people only have 2 extruders, but for those few that have many, there could just be more columns.  

Thanks, Steve

[image]