Ultimaker/Cura #7360: infill before walls

maukcc, 2020-03-27:
**Application version**
4.5 PPA master

**Platform**
Linux

**Printer**
HMS434

**Reproduction steps**
When you load a material profile settings that have "infill before walls" enabled, It will not get done until you manually disable and enable it again.

Ghostkeeper, 2020-03-30:
I'm unable to reproduce this. For example the Ultimaker S5 profile has this setting enabled by default, and it indeed prints the infill before the walls:

[image]

Similarly if I change the default in hfm434.def.json from `false` to `true` (on line 110):
[image]

No need to uncheck and check or anything.

Can you reproduce this by using the AppImage release? This might be something we've already fixed.

maukcc, 2020-04-01:
This happens when I load a package file with a quality file that has infill before walls

Ghostkeeper, 2020-04-02:
I'm still unable to reproduce this. These are now my reproduction steps:

* I made a .curapackage file: [InfillBeforeWallsTest.zip](https://github.com/Ultimaker/Cura/files/4421566/InfillBeforeWallsTest.zip). This package contains an arbitrary material profile that's not used by HMS434. It also contains a quality profile for HMS434 for the 0.4mm nozzle but quality type "coarse", since the HMS434 has a global quality for Coarse.
* Load in that package into Cura and reboot.
* Select the "Coarse" quality for HMS434, using any PLA material and the 0.4mm nozzle.
* Change the layer height to 0.1 so that it's accepted by the limits imposed by the definition.
* Verify that "Infill Before Walls" is indeed enabled by this profile.
* Slice a cube. Inspect layer view.
* On a layer halfway, slide through the horizontal scrollbar for the layer to verify that the infill is printed before the walls. Indeed, the infill is still printed before the walls.

Could you *please* provide a bit more extensive reproduction steps? Maybe the package that you're using and like a step-by-step guide of how you come to your conclusion? I'm putting in a lot of effort to help you but it doesn't feel like I get any guidance on how you're getting to this bug.

maukcc, 2020-04-08:
What we do
in printer definition file it is set to true
in package file it is set to false
Then for us it shows as true, but is actually false

also, is there an option do do this for the top and bottom walls(infill of walls) as well?

Ghostkeeper, 2020-04-09:
No, there are currently only settings for reordering infill vs. walls and inner vs. outer walls. The rest of the printing order is [fixed](https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_paths.md#printing-order).

Ghostkeeper, 2020-04-09:
With the defaults changed around I was able to reproduce this bug. Very weird! There is no inheritance function for it or whatever.

Ghostkeeper, 2020-04-09:
So the reproduction steps are then:
* In hms434.def.json, change the default for infill_before_walls to `true`.
* Start Cura.
* Load this [updated package file](https://github.com/Ultimaker/Cura/files/4457323/InfillBeforeWallsTest.zip) and reboot.
* Add an HMS434 printer and select the "Coarse" quality for PLA+0.4mm nozzle.
* FInd the Infill Before Walls setting. It's set to True, even though the profile sets it to False.
* In the God Mode plug-in it's revealed that it's indeed set to False for all extruders, only not for Global which is correct.

nallath, 2020-04-14:
So, I actually can't reproduce this. The package provided by @Ghostkeeper has a small mistake, since it sets the value to "false" instead of False. This means that instead of setting it as a boolean, it gets set as a string.
If you change the profile to use False, it does work. 

@maukcc Could you share the original package that has this issue? That way we can validate if it's really the issue I described.

maukcc, 2020-04-14:
Yep, that was it. It works with a capital F

closed 2020-04-14 by maukcc

maukcc, 2020-04-14:
one other thing (and this is not limited to infill_before_walls, but happens with other settings as well), if you do:

> infill_before_walls = =True if infill_sparse_density < 90 else False

and you change the sparse density it will not show the changed infill_before_walls, until you go to an other extruder and back.

Ghostkeeper, 2020-04-21:
Yeah we've seen that issue before. The problem is that we link setting updates to each other depending on which settings depend on each other according to the definition (i.e. the printer profile). So it doesn't know that it needs to update the setting. It's a known bug.