Ultimaker/Cura #7361: Useless support walls

michcior2, 2020-03-27:
**Application version**
4.5.0

**Platform**
Linux

**Printer**
N/A

**Reproduction steps**
In support settings:
1. Select "Support Placement = Touching Buildplate"
2. Set "Support Horizontal Expansion > 0 "

**Screenshot(s)**
[image]

**Actual results**
There are useless support constructions build around holes which are not to be supported at all  (not "Touching Buildplate" case)

mahtDFR, 2020-03-30:
Do you have 'support horizontal expansion' enabled under your support settings?

mahtDFR, 2020-03-30:
Thanks for the report. Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so we can try to reproduce it.

michcior2, 2020-03-30:
Here you are the project. Best Regards, Piotr. Temat: Re: [Ultimaker/Cura] Useless support walls (#7361)Data: 2020-03-30 15:34Nadawca: "maht" Adresat: "Ultimaker/Cura" ; DW: "michcior2" ; "Author" ; 
Thanks for the report. Please save a project file (File -&gt; Save...), zip up the .3mf file and attach to this issue so we can try to reproduce it.

mahtDFR, 2020-03-30:
I think you've replied via email, so your attachments haven't worked. Can you try again on GitHub itself please?

michcior2, 2020-03-30:
Here is the file, which I have found the bug. The profile settings are set to reproduce the bug. 
[CFFFP_issue7361.zip](https://github.com/Ultimaker/Cura/files/4404559/CFFFP_issue7361.zip)

Ghostkeeper, 2020-03-31:
This support is caused by having a large Support Horizontal Expansion. If the Support Horizontal Expansion exceeds the Support X/Y distance, the tiniest bits of overhang will cause big swabs of support. Those big swabs of support are then made useless because the support is not allowed to come close to your model due to the X/Y distance.

You do have tiny bits of overhang there:
[image]

I'd suggest adjusting one of these settings:
* Reduce Support Horizontal Expansion to be below Support X/Y Distance.
* Increase Support X/Y Distance to be above Support Horizontal Expansion.
* Use the Make Overhang Printable, adjust the Maximum Model Angle to 45 (your Overhang Angle setting) and disable support altogether.

And thanks for helping out the effort, by the way! I recognise that model.

closed 2020-03-31 by Ghostkeeper