Ultimaker/Cura #7362: Using Da Vinci Jr. Pro with Ultimaker Cura

landonbarnett4, 2020-03-27:
**Application version**
4.5.0

**Platform**
MacOS and Windows 10. 

**Printer**
Custom FFF printer (da Vinci Jr. Pro)

**Reproduction steps**
1. I configured the Printer Settings under the Printer tab. 
2. I tried to configure the Printhead Settings under the Printer tab. 

**Screenshot(s)**
[image] or https://imgur.com/Xo7UYCl

**Actual results**
After I enter the correct settings that are available on the XYZware website, the Printhead Settings correct to different numbers than I have inputted. 

**Expected results**
I expected the Printhead Settings to be perfectly fine and stay as I input them.

**Project file**
There was no project file imported at this time.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4395765/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4395766/DxDiag.txt)

**Additional information**
None.

nallath, 2020-03-30:
The nozzle is assumed to be at (0,0). As such, the min point of the head size _must_ be negative (or zero) and the max size _must_ be positive (or zero).

As such, this is intentional behavior.

closed 2020-03-30 by nallath

Ghostkeeper, 2020-03-31:
No support yet for quantum tunneling filament that teleports outside the print head as soon as it's deposited? :cry:

mottainai-x, 2021-04-08:
I've come up with the same problem on XYZ - Da Vinci Nano. 

What should the Printhead Settings be as the XYZ website X min and Y min as positive (10mm)?
https://xyzofficialweb.blob.core.windows.net/xyz-v3/download/Cura_Profile_XYZ.PDF

Ghostkeeper, 2021-04-15:
@mottainai-x Change X Min to `-10` and Y Min to `-10`. The rest can be left as-is.