Ultimaker/Cura #7364: [4.5.0] window zoom issue

Brendan-Loy, 2020-03-28:
**Application version**
4.5.0

**Platform**
Windows 10, surface pro 6

**Printer**
Artillery Sidewinder X1

**Reproduction steps**
i just updated from 4.4.0 to 4.5.0 and running the new version of cura for the first time everything worked fine, i sliced and printed a couple things. But now wherever i launch it the window is about twice the size it should be and everything is super zoomed in, its so bad i cant even see the build plate.

i uninstalled and reinstalled it and the exact same thing happend, worked the first time but then went super zoom from then on

if i full screen the window everything remains zoomed in 
**Screenshot(s)**
[image]
this is what it looks like immediately upon opening, note the window options (close minimize etc.) arent visible in the top right untill i drag the window to the left (shown below) just to give a sense of how wide the window is
[image]

this is what it looks like if you put the window in fullscreen
[image]

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4396537/cura.log)

Brendan-Loy, 2020-03-28:
update: running in windows 8 compatibility mode seems to work but only when set up using the windows troubleshooter, i turned it on myself on the actual .exe and the problem stayed

Ghostkeeper, 2020-03-31:
Could you try removing the qtshadercache folder from `%APPDATA%/local`? You have some symptoms of a corrupt Qt shader cache.

GregValiant, 2026-05-30:
Is this still a problem in 5.12.1 or can this be closed?