Ultimaker/Cura #7365: [4.5.0] Curious skin fill order

Asterchades, 2020-03-28:
**Application version**
First observed with 4.5.0. Unsure if occurred earlier.

**Platform**
Win10 x64 Pro 1909
AMD Ryzen 1700 (@3.7GHz)
GeForce GTX1080 (v442.74)

**Printer**
Modified Creality Ender-2.
Custom profile based on default FDM:
[Thalyn_FDM.def.zip](https://github.com/Ultimaker/Cura/files/4396806/Thalyn_FDM.def.zip)

**Reproduction steps**
1. Slice model.
2. Observe weird behaviour during print with regards to the order skin lines are printed on some of the sloped walls.

**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
As per the above screenshots, one side of the skin is produced with steps between the extrusions. This behaviour occurs on multiple layers, and always on the same side relative to the extrusion direction. By that I mean the next layer up has it occur on the longer far side as the extrusions are going in the alternative direction, two layers up has the same side affected as this one, and so forth.

**Expected results**
Ideally I would imagine that the extrusions would print side-by-side, eliminating a large amount of unnecessary travels, as it does on the other, unaffected side and the two shorter sides. This is not the first time I've seen this sort of awkwardness in the extrusion pattern but it is the most repeatable and imposes the largest travel penalty.

**Project file**
[Cholo - Rat.zip](https://github.com/Ultimaker/Cura/files/4396798/Cholo.-.Rat.zip)

0.2mm layer height, 3 walls, starting from layer 7, continuing to least layer 41.

**Log file**
N/A as no crash or other misbehaviour occurred.

**Additional information**
If you're at all curious, this is a first attempt at modelling the Ratdroid from the 1986 Firebird game "Cholo", as seen in-game on the C64.

smartavionics, 2020-03-28:
This is a known limitation of the currently implementation. You should get better results using zigzag rather than lines skin pattern.

Asterchades, 2020-03-28:
What an odd limitation. From the outside looking in, I would have imagined the only difference between Zig-Zag and Lines for the skin would be the presence of some additional retractions for Lines when not going exactly on the angle.

Is there some reason I'm not seeing that the two don't share basic code paths? It feels like an unnecesary duplication of effort.

smartavionics, 2020-03-28:
Both the lines and zigzag patterns are printed using the same code. In both cases, after printing a line it looks to find the nearest line not yet printed and so on. When the skin lines are quite short (the skin region is narrow) and the skin lines meet the wall lines at a very shallow angle, you can get the effect you are seeing. The zigzag skin pattern doesn't tend to have that issue because the zags (or are they the zigs?) that join the ends of the skin lines are always closer than the ends of the skin lines on the other side of the skin.

Asterchades, 2020-03-28:
Ah, I think I get what you're saying.

Wouldn't this have to mean that the pathfinder is incorrectly evaluating the shortest path by not testing both ends of a given extrusion? The only way that those skipped lines could be considered "optimal" would be if the pathfinder were mistakenly believing that the furthest end of the adjacent line was where it was going to start, or it was only given that one point to make a determination.

smartavionics, 2020-03-29:
When looking for the next line to print, it does choose the line end that is closest, it checks both ends of the remaining lines. But when the skin region is narrow and the lines are hitting the boundary at a shallow angle, the closest line end can be on the other side of the skin area rather than the line end that is next to the line end that has just been reached. This is more likely to happen when using wide skin lines.

Asterchades, 2020-03-29:
So, in essence then, all it's really doing is making the print order look weird (and possibly reducing extrusion-to-extrusion adhesion slightly by virtue of the older lines being permitted to cool more).  It genuinely is picking the shortest travel path, but the acute angle involved means the adjacent extrusion actually starts relevantly further away.

I am, however, going to offer one last image which might explain a little bit of my confusion. This is taken from layer 14 of the model embedded above with the camera up as high as possible and the affected area as close to the middle of the screen as possible to remove any perspective error:
[image]

Both circles have a radius of 86 pixels. The blue circle is centred where the travel begins, while the purple circle is centred where the combing travel enters the wall. Neither one shows the further extrusion being closer, at least as far as the end of the travel movements is concerned. Experimenting with combing settings suggests the blue circle is the one that matters.

The only thing I can see from that image is that the travels are overshooting the centre of the extrusion start. This would either be a display error on the travel or an additional, obscured (by the existing travel indication) travel movement, but could explain it as the far extrusion definitely has its centre closer than the near one. Or is this perhaps just a case of the distances being so extremely tiny (we're talking micrometres) that it's being rounded/truncated because it makes no appreciable difference in practice?

_I get that, seeing the above image, this is all just basically splitting hairs. The extra distance - if there is any - is so tiny it may as well not exist. I'm just curious at this point, and I'll understand if that's not enough to continue (hence also closing the issue)._

closed 2020-03-29 by Asterchades

smartavionics, 2020-03-29:
Yes, you understand correctly what's happening. As for the little hook shaped combed travels, they are due to the fact that the combing crosses the skin to the nearest combing boundary, travels along the boundary a short distance and then goes to the start of the next line. It's suboptimal to be sure but whether it actually makes much difference is debatable. Cheers!