Ultimaker/Cura #7366: Improve "Fill Gaps Between Walls"

michcior2, 2020-03-28:
**Is your feature request related to a problem? Please describe.**
Setting up the feature "Fill Gaps Between Walls" sometimes is necessary, because that's the only way to fill some areas:
[image]

Unfortunately, it generates huge number of small particles, which waste a lot of time for travel, retract etc in some specific cases: sharp inner corners. See my case (file attached):
[image]

**Describe the solution you'd like**
There is need for some kind "Filter small particles" or "Minimum extrusion length" etc.

[CFFFP_fillgapsissue.zip](https://github.com/Ultimaker/Cura/files/4396947/CFFFP_fillgapsissue.zip)

smartavionics, 2020-03-28:
Until UM provide a better gap filling implementation (which is being worked on, I believe) you can use my Cura builds which do this differently and make a better job of it. You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there before using.

Ghostkeeper, 2020-03-31:
> There is need for some kind "Filter small particles" or "Minimum extrusion length" etc.

You mean [Filter Out Tiny Gaps](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/filter_out_tiny_gaps.md)?

closed 2020-12-01 by Ghostkeeper