Ultimaker/Cura #7367: Fan speed control for ironing and overhangs

wimiam1, 2020-03-28:
**Is your feature request related to a problem? Please describe.**
I've noticed there are a multitude of ironing and overhang specific print settings, but neither includes fan speed, which is probably much more useful than print speed in the case of overhangs. I'd also appreciated a fan speed setting for ironing as sometimes I want no part cooling for the sake of layer adhesion, but having the fan off for ironing can lead to a very melted-appearing surface.

**Describe the solution you'd like**
Just a fan speed percentage text box for ironing and overhanging walls.

**Describe alternatives you've considered**
I've resorted to using the ChangAtZ plugin, but this applies the fan speed to the whole layer which can be detrimental to layer adhesion and warping.
**Affected users and/or printers**
While I don't use ABS or other warping-prone materials at the time, I'm sure separate fan control for overhanging walls would help to improve overhang performance without affecting the rest of the print. I also would find it helpful for printing flexible objects that need to have a long lifetime. You need the best layer adhesion possible to ensure flexing won't cause the layers to separate, but low fan speed during ironing causes severe melting on the top surfaces.

smartavionics, 2020-03-28:
Hi, my cura builds feature an overhanging wall fan speed setting. They are available from https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there for more info.

Ghostkeeper, 2020-03-31:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/5140.

closed 2020-03-31 by Ghostkeeper