Ultimaker/Cura #7369: Machine setings do not save.

Stormwing01, 2020-03-29:
**Application version**
4.5

**Platform**
windows 10 pro 64 bit

**Printer**
wanhao i3 mini

**Reproduction steps**
1. go to settings -> printer -> manage printer ->machine settings -> extruder 1

If I change the nozzle diameter from 0.4 to 0.3 and the filament diameter from 1.75 to 1.73, they both reset as soon as I click close and exit to use cura. when I go back to that setting, they are back to 0.4 and 1.75 respectively.

**Screenshot(s)**
this is not really something I think I can screenshot as it's not a graphical issue

**Actual results**
nozzle and filament diamter settings in "machine settings" are not saved.

**Expected results**
the software should have saved those settings and made adjustments accordingly, because I do not use a 0.4mm nozzle, and my filament diameter is at this moment an average of 1.73mm. it impacts print quality negatively.

**Project file**
there is no project.

**Log file**
I have no cura.log file.

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-03-29:
Could you try entering the numbers with a comma instead of a dot?

Stormwing01, 2020-03-29:
edit: when I save te settings as with a "," and not a "." in the value, they do save, but if I simply edit and exit with a "." in the value, it resets itself....

fieldOfView, 2020-03-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/6546

closed 2020-03-30 by mahtDFR