Ultimaker/Cura #7372: Ultimaker Cura 4.5 Issues

CTotten-Sonalysts, 2020-03-30:
I installed Cura 4.5 about 3 weeks ago (previously was using Cura 4.4.1).  Since the new installation, I have been having some odd issues.
Everytime I open Cura, it seems to think it is the first time I have opened it.  I get the "Welcome" screen, where I have to agree to the Terms and Conditions, and it goes through the setup process, including adding networked printers.
Once I get into Cura, I keep getting a message saying my configuration is corrupt (see screen shot attached).
[image]
I went through the reset process that it recommended, and it seemed to work OK for a day or so, then it went back to the same issues.  Cura log attached below.
[cura.log](https://github.com/Ultimaker/Cura/files/4403750/cura.log)

I have an Ultimaker S5 and 3E connected via ethernet.  The printers seem to be working just fine, printing items from the Cura Connect queue, but the something definitely does not seem right with Cura.  The welcome screen is annoying, especially when you see it daily.  And I am tired of having to restore my tuned settings.

mahtDFR, 2020-03-31:
Thanks for the report. If you're getting the welcome flow at each startup, that would indicate your [user data](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) isn't getting saved.

You didn't specify if you're on Windows but I'm assuming you are. Try this:

- Go to `%APPDATA%\cura\`
- Zip up your 4.4.1 and 4.5 configuration folders to make a backup
- Delete your existing user data
- Start 4.5

Does it start cleanly? Does the welcome flow at startup behaviour persist?

Some relevant log output is here:

`2020-03-30 07:26:02,703 - ERROR - [MainThread] UM.Settings.ContainerStack.deserialize [396]: When trying to deserialize ultimaker_s5_extruder_left #3, we received an unknown container ID (ultimaker_s5_extruder_left #3_user)
2020-03-30 07:26:02,705 - ERROR - [MainThread] UM.Logger.logException [98]: Exception: Error in the format of container ultimaker_s5_extruder_left #3: When trying to deserialize ultimaker_s5_extruder_left #3, we received an unknown container ID (ultimaker_s5_extruder_left #3_user)
2020-03-30 07:26:02,707 - ERROR - [MainThread] UM.Logger.logException [102]: Traceback (most recent call last):
2020-03-30 07:26:02,708 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 183, in findContainers
2020-03-30 07:26:02,710 - ERROR - [MainThread] UM.Logger.logException [102]:   File "C:\Program Files\Ultimaker Cura 4.5\plugins\LocalContainerProvider\LocalContainerProvider.py", line 68, in loadContainer
2020-03-30 07:26:02,711 - ERROR - [MainThread] UM.Logger.logException [102]:     container.deserialize(f.read(), file_path)
2020-03-30 07:26:02,713 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\cura\Settings\ExtruderStack.py", line 165, in deserialize
2020-03-30 07:26:02,715 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerStack.py", line 259, in deserialize
2020-03-30 07:26:02,717 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerStack.py", line 397, in deserialize
2020-03-30 07:26:02,719 - ERROR - [MainThread] UM.Logger.logException [102]: UM.Settings.ContainerFormatError.ContainerFormatError: When trying to deserialize ultimaker_s5_extruder_left #3, we received an unknown container ID (ultimaker_s5_extruder_left #3_user)
2020-03-30 07:26:02,722 - ERROR - [MainThread] UM.Settings.ContainerStack.deserialize [396]: When trying to deserialize ultimaker3_extended_sync, we received an unknown container ID (Thor_user)
2020-03-30 07:26:02,724 - ERROR - [MainThread] UM.Logger.logException [98]: Exception: Error in the format of container ultimaker3_extended_sync: When trying to deserialize ultimaker3_extended_sync, we received an unknown container ID (Thor_user)
2020-03-30 07:26:02,726 - ERROR - [MainThread] UM.Logger.logException [102]: Traceback (most recent call last):
2020-03-30 07:26:02,728 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 183, in findContainers
2020-03-30 07:26:02,730 - ERROR - [MainThread] UM.Logger.logException [102]:   File "C:\Program Files\Ultimaker Cura 4.5\plugins\LocalContainerProvider\LocalContainerProvider.py", line 68, in loadContainer
2020-03-30 07:26:02,732 - ERROR - [MainThread] UM.Logger.logException [102]:     container.deserialize(f.read(), file_path)
2020-03-30 07:26:02,733 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerStack.py", line 259, in deserialize
2020-03-30 07:26:02,735 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerStack.py", line 397, in deserialize
2020-03-30 07:26:02,737 - ERROR - [MainThread] UM.Logger.logException [102]: UM.Settings.ContainerFormatError.ContainerFormatError: When trying to deserialize ultimaker3_extended_sync, we received an unknown container ID (Thor_user)
2020-03-30 07:26:02,744 - ERROR - [MainThread] UM.Settings.ContainerStack.deserialize [396]: When trying to deserialize Thor #4, we received an unknown container ID (Thor #4_user)
2020-03-30 07:26:02,746 - ERROR - [MainThread] UM.Logger.logException [98]: Exception: Error in the format of container Thor #4: When trying to deserialize Thor #4, we received an unknown container ID (Thor #4_user)
2020-03-30 07:26:02,748 - ERROR - [MainThread] UM.Logger.logException [102]: Traceback (most recent call last):
2020-03-30 07:26:02,750 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 183, in findContainers
2020-03-30 07:26:02,752 - ERROR - [MainThread] UM.Logger.logException [102]:   File "C:\Program Files\Ultimaker Cura 4.5\plugins\LocalContainerProvider\LocalContainerProvider.py", line 68, in loadContainer
2020-03-30 07:26:02,754 - ERROR - [MainThread] UM.Logger.logException [102]:     container.deserialize(f.read(), file_path)
2020-03-30 07:26:02,756 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerStack.py", line 259, in deserialize
[... cut ...]
2020-03-30 07:26:02,858 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerStack.py", line 259, in deserialize
2020-03-30 07:26:02,860 - ERROR - [MainThread] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerStack.py", line 397, in deserialize
2020-03-30 07:26:02,861 - ERROR - [MainThread] UM.Logger.logException [102]: UM.Settings.ContainerFormatError.ContainerFormatError: When trying to deserialize ultimaker_s5_extruder_left #4, we received an unknown container ID (ultimaker_s5_extruder_left #4_user)`

CTotten-Sonalysts, 2020-03-31:
I am on Windows 10.  And just do be clear, under AppData\Roaming\cura\ I have 3 folders (4.3, 4.4, and 4.5).  Are these the configuration folders you are talking about?  If so, I should probably delete the oldest ones (4.3)?  And I guess my previous version was 4.4, not 4.4.1.
 
Chuck
 
Charles Totten
Operations Analyst
Sonalysts, Inc.
(860) 326-3810

CTotten-Sonalysts, 2020-03-31:
I zipped my configuration folders for all 3 installations on the machine (4.3, 4.4, and 4.5).  I then deleted those folders and opened Cura.  I went through the Welcome screen process (expected), and set up a print file.  All worked as expected.  I closed Cura and reopened it.  I did not get the Welcome screen this time, or the corrupt profiles warning.  I set up another print file and sent it to the printers.  I closed Cura again, and reopened without seeing the Welcome screen or the corrupt profile warning.
 
I think this fixed the problem.  Thanks for the help.  
 
In the future, when I upgrade to a newer version of Cura, should I make a point of deleting old configuration files?
 
Regards,
Chuck
 
Charles Totten
Operations Analyst
Sonalysts, Inc.
(860) 326-3810

mahtDFR, 2020-03-31:
> I think this fixed the problem.  Thanks for the help. 

This is more of a workaround than a fix, but glad to hear its got you in a position where you can use it again.

> In the future, when I upgrade to a newer version of Cura, should I make a point of deleting old configuration files?

You shouldn't need to do this at all, there are scripts in place that are supposed to carry over your settings from a previous installation. This time we just did it to verify if there was an issue with your user settings ,which seems to be the case. Something in there is incompatible, but I'm not sure what. Perhaps someone else reading this has ideas. Have you installed any machine definitions or plugins manually? (i.e. by dragging and dropping files into your installation)

CTotten-Sonalysts, 2020-03-31:
I have not installed any plugins.  I am not sure what you mean by machine definitions.  If you are referring to the printers installed, I have had some issues with that, mainly when I have to cycle power to a machine, since that changes to the IP address (since Ultimaker still has not provided for fixed IP addresses).
 
Charles Totten
Operations Analyst
Sonalysts, Inc.
(860) 326-3810

rburema, 2020-03-31:
Hopefully this should (at least for part of the cases) be fixed for the upcoming 4.6, see commit https://github.com/Ultimaker/Cura/pull/7356/commits/3c50b78e685c1c226db654852c03a98ee8f7311b

I'll close this as resolved for now, please open a new issue if this persists in the future new version.

closed 2020-03-31 by rburema