Ultimaker/Cura #7379: Cura 4.5 Anet profile issue

PaddoSwam, 2020-03-31:
**Application version**
Cura 4.5

**Platform**
Windows

**Printer**
Anet A8 (but applies to all profiles that derive from Anet)

**Reproduction steps**
1. Default installation
2. Select Anet A8
3. Select a model (benchy for example)
4. Slice
5. Preview
6. See infill wipe protrude through shell

**Screenshot(s)**
[image]

**Actual results**
Artifacts show up and also show up while printing.

**Expected results**
[image]

tycho94, 2020-03-31:
Just for info, this is also present in the A6

nallath, 2020-04-01:
This is most likely caused by some settings in the profiles. These profiles have been contributed to us by others. Since we don't have any Anet printers (or have the time to actually test this), we don't validate any of these profiles outside of checking if they crash Cura.

tycho94, 2020-04-01:
I understand, I was not aware this setting was global (for Anet printers), thus thought it had to be editted for multiple seperate printer configs. However I saw in the patch it is done for all printers. 

I can confirm for you that it is fixed for the Anet A6.

rburema, 2020-04-01:
@nallath See the PR ( #7380 ) linked to above by the original poster, it's supposed to fix it, for other Anet printers as well, since it changes the base.

PaddoSwam, 2020-04-01:
@tycho94 you're right and I've created a fix #7380  for all Anet based profiles like @rburema mentioned

rburema, 2020-04-02:
Cherry picked the fix into the 4.6-branch instead, with [this commit](https://github.com/Ultimaker/Cura/commit/f3927115d6e0cc2a45de2507f64c8c4bdafe08f8).

Note for others: Please do _not_ take this as a sign to merge to a version branch. In general _always_ base on and merge to master. "I really like this feature/fix to be in there" is not a good enough reason. I only do this because it was a one line fix in a 3rd party profile _which do not require testing on our side_ & reserve the right to _not_ do this at other times.

(Devs: See internal ticket CURA-7337)

closed 2020-04-02 by rburema