Ultimaker/Cura #7381: [4.5.0] Hole for bolt-nut slicing error

andrewpho, 2020-03-31:
**Application version**
4.5.0

**Platform**

Mac OSX 10.15.3

**Printer**
AnyCubic 4Max Pro

**Reproduction steps**
1. import screw nut STL into Cura
[image]

2. when placed flat on print-bed and sliced, the hole is not properly sliced and is covered over
[image]

3. when placed vertically on print-bed, the hole is properly sliced.
[image]
[image]
 

**Actual results**
hole is filled with material

**Expected results**
hole should be free of material

**Project file**

[nut.3mf.zip](https://github.com/Ultimaker/Cura/files/4410957/nut.3mf.zip)

**Log file**
2020-03-31 11:43:32,483 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] M140 S0 ; turn off bed
2020-03-31 11:43:32,487 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] M84 ; disable motors
2020-03-31 11:43:32,491 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] M107
2020-03-31 11:43:32,495 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] G91 ;relative positioning
2020-03-31 11:43:32,499 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
2020-03-31 11:43:32,504 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] G1 Z+0.5 E-5 ;X-20 Y-20 F60 ;move Z up a bit and retract filament even more
2020-03-31 11:43:32,508 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
2020-03-31 11:43:32,512 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] G1 Y180 F2000
2020-03-31 11:43:32,516 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] M84 ;steppers off
2020-03-31 11:43:32,520 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] G90
2020-03-31 11:43:32,524 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] M300 P300 S4000" -s machine_center_is_zero="False" -s jerk_roofing="3" -s retraction_min_travel="0.8" -s raft_base_line_width="0.8" -s wall_0_extruder_nr="-1" -s support_infill_extruder_nr="0" -s speed_prime_tower="40" -e0 -s speed_support_interface="29" -s extruder_prime_pos_y="0" -s jerk_support_interface="3" -s support_roof_density="100" -s cool_min_layer_time_fan_speed_max="10" -s bridge_wall_material_flow="50" -s raft_base_jerk="11" -s support_tree_angle="40" -s support_interface_skip_height="0.3" -s bottom_skin_preshrink="1.2000000000000002" -s top_bottom_pattern="zigzag" -s small_feature_speed_factor="50" -s speed_print="50" -s support_roof_offset="0.0" -s jerk_layer_0="3" -s machine_steps_per_mm_z="50" -s extruder_prime_pos_x="0" -s layer_start_y="0.0" -s bridge_sparse_infill_max_density="0" -s raft_acceleration="900" -s material_print_temperature="200" -s infill_enable_travel_optimization="False" -s prime_blob_enable="False" -s ironing_only_highest_layer="False" -s raft_surface_thickness="0.2" -s speed_equalize_flow_max="150" -s mesh_position_y="0" -s bridge_skin_density="100" -s support_initial_layer_line_distance="8.0" -s skirt_brim_minimal_length="250" -s raft_surface_layers="2" -s initial_layer_line_width_factor="100.0" -s z_seam_position="back" -s jerk_support_roof="3" -s spaghetti_infill_stepped="True" -s switch_extruder_retraction_speeds="20" -s material_guid="0ff92885-617b-4144-a03c-9989872454bc" -s outer_inset_first="False" -s resolution="0" -s z_seam_y="205" -s acceleration_layer_0="300" -s top_bottom_pattern_0="zigzag" -s cool_min_speed="10" -s gradual_infill_step_height="1.5" -s retraction_combing_max_distance="0" -s z_seam_x="135.0" -s infill_offset_y="0" -s ironing_line_spacing="0.1" -s acceleration_support="450" -s infill_multiplier="1" -s support_connect_zigzags="True" -s spaghetti_infill_extra_volume="0" -s skin_outline_count="1" -s jerk_prime_tower="7" -s support_offset="0" -s material_flow="100" -s retraction_hop_after_extruder_switch_height="1" -s machine_extruder_end_pos_abs="False" -s material_end_of_filament_purge_length="20" -s acceleration_support_bottom="300" -s material_break_preparation_speed="2" -s material_break_temperature="50" -s wipe_retraction_amount="6.5" -s max_skin_angle_for_expansion="90" -s connect_infill_polygons="False" -s machine_extruder_start_code="" -s infill_wipe_dist="0.1" -s acceleration_skirt_brim="300" -s meshfix="0" -s prime_tower_min_volume="6" -s cool_lift_head="False" -s speed_wall_0="29" -s acceleration_wall_x="300" -s skin_material_flow="100" -s support_bottom_density="100" -s support_bottom_height="1" -s bridge_skin_material_flow_2="100" -s speed="0" -s support_roof_pattern="concentric" -s wipe_hop_amount="1" -s support_join_distance="2.0" -s mold_angle="40" -s raft_speed="25.0" -s acceleration_support_infill="450" -s brim_gap="0" -s xy_offset="0" -s infill_overlap="10" -s extruder_nr="0" -s support_interface_material_flow="100" -s material_flush_purge_length="60" -s machine_nozzle_heat_up_speed="2.0" -s raft_interface_line_spacing="1.0" -s support_material_flow="100" -s bridge_fan_speed="100" -s travel="0" -s support_skip_some_zags="False" -s speed_infill="50" -s skin_overlap="10" -s top_thickness="1.2" -s raft_interface_jerk="11" -s material_shrinkage_percentage="0" -s machine_nozzle_size="0.4" -s roofing_pattern="zigzag" -s support_bottom_offset="0.0" -s bridge_wall_speed="14.5" -s meshfix_union_all="True" -s support_z_distance="0.1" -s machine_endstop_positive_direction_y="False" -s minimum_bottom_area="1.0" -s infill_line_distance="0" -s support_top_distance="0.1" -s retraction_hop_after_extruder_switch="True" -s retraction_extra_prime_amount="0" -s support_bottom_line_distance="0.4" -s prime_tower_wipe_enabled="True" -s fill_perimeter_gaps="everywhere" -s jerk_print_layer_0="3" -s material_anti_ooze_retracted_position="-4" -s wipe_hop_enable="False" -s zig_zaggify_support="False" -s meshfix_keep_open_polygons="False" -s magic_fuzzy_skin_enabled="False" -s max_extrusion_before_wipe="10" -s raft_base_line_spacing="1.6" -s speed_support_infill="29" -s machine_endstop_positive_direction_z="True" -s coasting_volume="0.064" -s layer_start_x="0.0" -s machine_min_cool_heat_time_window="50.0" -s switch_extruder_retraction_speed="20" -s retraction_hop_enabled="False" -s acceleration_roofing="300" -s spaghetti_inset="0.2" -s support_use_towers="True" -s acceleration_infill="900" -s travel_avoid_supports="False" -s speed_z_hop="10" -s gradual_infill_steps="4" -s material_end_of_filament_purge_speed="0.5" -s raft_margin="15" -s wall_line_width_0="0.4" -s cross_infill_pocket_size="0" -s connect_skin_polygons="False" -s infill_pattern="quarter_cubic" -s support_infill_angles="[ ]" -s shell="0" -s support_tower_diameter="3.0" -s machine_extruder_start_pos_x="0" -s sub_div_rad_add="0.4" -s support_interface_height="1" -s acceleration_wall="300" -s acceleration_travel="900" -s ironing_enabled="False" -s support_bottom_material_flow="100" -s skin_angles="[ ]" -s machine_extruder_end_pos_x="0" -s acceleration_support_interface="300" -s spaghetti_max_infill_angle="10" -s meshfix_extensive_stitching="False" -s magic_fuzzy_skin_thickness="0.3" -s mold_roof_height="0.5" -s machine_filament_park_distance="16" -s fill_outline_gaps="False" -s support_zag_skip_count="2" -s raft_interface_thickness="0.30000000000000004" -s roofing_angles="[ ]" -s jerk_ironing="3" -s skin_edge_support_layers="0" -s travel_compensate_overlapping_walls_x_enabled="True" -s blackmagic="0" -s infill_support_enabled="False" -s cool_fan_speed_0="0" -s support_brim_line_count="20" -s cool_fan_speed_max="100.0" -s cool_fan_enabled="True" -s wall_overhang_angle="90" -s raft_surface_line_width="0.4" -s wall_min_flow="0" -s ironing_flow="10.0" -s machine_steps_per_mm_y="50" -s support_tree_branch_diameter="2" -s speed_wall="43" -s support_xy_distance="0.7" -s machine_steps_per_mm_x="50" -s wall_line_count="3" -s magic_fuzzy_skin_point_density="1.25" -s raft_jerk="11" -s speed_print_layer_0="20" -s support_line_distance="8.0" -s retraction_enable="True" -s z_seam_type="sharpest_corner" -s retraction_prime_speed="25" -s wall_0_material_flow="100" -s skin_no_small_gaps_heuristic="False" -s support_line_width="0.4" -s speed_topbottom="29" -s raft_smoothing="5" -s wipe_retraction_retract_speed="25" -s multiple_mesh_overlap="0.15" -s bridge_skin_material_flow_3="110" -s initial_bottom_layers="3" -s bridge_fan_speed_2="0" -s support_ [... cut ...]
2020-03-31 11:43:32,920 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [183]: Socket state changed to Closed
2020-03-31 11:43:32,921 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [829]: Backend quit with return code 0. Resetting process and socket.
2020-03-31 11:43:33,775 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles

**Additional information**
project file contains a second STL that works properly, even when placed flat on print-bed. This is a very similar nut with similar hole.

andrewpho, 2020-03-31:
[nut that does not work.stl.zip](https://github.com/Ultimaker/Cura/files/4411035/nut.that.does.not.work.stl.zip)
[nut that works.stl.zip](https://github.com/Ultimaker/Cura/files/4411037/nut.that.works.stl.zip)

andrewpho, 2020-03-31:
"Remove All Holes" is not checked

smartavionics, 2020-03-31:
This is what mesh tools says about nut.3mf...

[image]

andrewpho, 2020-03-31:
Thank you!
Both are not water tight per Mesh tools. When examined by x-ray view, there is visible red areas for the one that did not work. For the model that was successfully sliced, no red area was visible in x-ray view but I am able to "repair" the holes with Mesh Tools.

closed 2020-03-31 by andrewpho