Ultimaker/Cura #7382: G1 FXXX ZXXX is added in the top of Gcode when using "Z Hop when retracted"

Moezregayeg, 2020-04-01:
This is a G code generating issue with Ultimaker 2 printer, Cura 4.4 and 4.5
Below first lines of Gcode generated.
Cura adds A line with G1 Fxxx Zxxx at the beginning of file (see line 21)
This cause the extruder to take the initial extruded material to thr printing area.
when removing "Z Hop when retracted", the problem disappears.
This is an extarct from the file.

;FLAVOR:UltiGCode
;TIME:9071
;MATERIAL:12298
;MATERIAL2:0
;NOZZLE_DIAMETER:0.4
;MINX:51.61
;MINY:126.915
;MINZ:0.3
;MAXX:160.94
;MAXY:199.308
;MAXZ:4.325
;Generated with Cura_SteamEngine 4.4.1
M82 ;absolute extrusion mode

G92 E0
G92 E0
G10
;LAYER_COUNT:18
;LAYER:0
M107
G1 F600 Z0.925
;MESH:rondelle.stl(1)
G0 F3600 X79.523 Y134.054 Z0.925

Ghostkeeper, 2020-04-02:
> This cause the extruder to take the initial extruded material to thr printing area.

What do you mean with this? The Ultimaker 2 doesn't purge before its print as far as I know. You'd need to use a skirt or something to prime the material.

Moezregayeg, 2020-04-02:
Hello,
Yes the Ultimaker 2 makes a purge before it prints.
I installed the firmware included in the 4.4 version of Cura.

darkdvd, 2020-05-06:
[youtube.com/watch?v=hWpaqMkcYq8](url)

First part of the video : object sliced without "Z hop when retracted"
Second part of the video : same object sliced WITH "Z hop when retracted"

CURA 4.6.1

Ghostkeeper, 2020-05-11:
Ah, that's right. It did make the little spiral purge. It's not initiated by Cura (e.g. like for the UM3 and up, where you can choose the purge location from Cura)

I'll see what we can do. Maybe it's best to just first do a quick horizontal move in the start g-code. It's a bit hard to test now since we're all working from home due to COVID-19 and while we all have printers at home, I don't think anyone has an UM2 any more. But I'm sure we can ask around if some hardware engineer can test it for us.

Moezregayeg, 2020-05-11:
Hi,
Tou can just not generate the line containing the M order.
This line is generated only if retract is active!

Ghostkeeper, 2020-05-13:
We're going to test adding this to the start g-code: `G0 F3000 Y50`

GregValiant, 2024-12-15:
Is this still an issue in current Cura versions (5.8 and up)?  Can this be closed?

closed 2024-12-30 by github-actions[bot]