Ultimaker/Cura #7383: GUI: Snap to grid when placing models

loneFunction, 2020-04-01:
**Is your feature request related to a problem? Please describe.**
When placing and moving models up to 4 decimal places are used for location. This extra precision is rarely required and quite irritating when placing models accurately. For every model a manual edit of the location field in necessary. CAD software almost always has an option to snap to a grid. There is even a visible grid in the display window, yet it does not serve a function except as visual estimation

**Describe the solution you'd like**
Snap to grid option or setting to round to nearest whole unit.

**Describe alternatives you've considered**
Snap is about as basic a feature as you can get.

**Affected users and/or printers**
Everyone, especially those who like seeing whole numbers.

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ghostkeeper, 2020-04-02:
Is there a reason why you'd need to place them in a very particular position, rounded to millimetres? For precise placement I'm indeed having to resort to using the text boxes, but having it snap to a grid of whole mm doesn't usually help since the models aren't always aligned to mm in real life either, as they have curves, and if they are there's usually something like support or a brim that destroys that anyway.

loneFunction, 2020-04-02:
A lot of what I make are mechanical parts without many complex curves. Being able to snap would allow placement with known spaces between them.
There are many instances where the location of a set object will change with things like rotate. It will be at 0,0 then rotate 90 and it's at 0,0.0001.
If snap were implemented things like nudge would also be able to added. When filling up a dense build plate being able to nudge objects 1mm at a time would be very time saving instead of grabbing handle, seeing what fits, select box and round to nearest whole.

My question is why not round to nearest whole numbers?

mahtDFR, 2020-04-06:
Hey there. Sorry to be the bearer of bad news but we've decided we're not going to take this up. We don't see a need to create a feature that snaps to the grid in a rounded position when you can adjust it precisely already using the positioning tools.

We're also working on refactoring and improving the 'multiple models on the build plate' code, so we wont be able to put time into this.

closed 2020-04-06 by mahtDFR

padcom, 2020-11-02:
You know,... Cura is such a good slicer and such a terrible model organizer. Snap to grid is a must-have. Please reconsider.

Ghostkeeper, 2020-11-04:
As a workaround, you can use an application that is a better model organizer too, save the result as 3MF and load that into Cura.

stotler, 2022-02-16:
Bummer, snap-to-grid seems like a no-brainer to me when placing a lot of options on the build plate.