Ultimaker/Cura #7386: Print Stops

mabus5, 2020-04-01:
**Application version**
4.5.0

**Platform**
Windows 10

**Printer**
Ender 3 Pro
[Test Files.zip](https://github.com/Ultimaker/Cura/files/4415841/Test.Files.zip)

**Reproduction steps**
1. Printed model from SD card
2. Model failed
3. Reset printer
4. Printed model again, from SD card
5. Model failed

**Actual results**
The print head stops moving while in contact with the print. The heaters stay on as well. The print head stops at around the middle of the right side of the model, near the "Y" letters if I recall correctly. It happens sometime after layer 51, but before the next change at z. The LCD updates at each change at z layer height and it's stopped in the same location both times, with the change at layer 51 message on the LCD. 

**Expected results**
The model should print.

**Project file**
Attached the gcode and 3mf files. 

**Additional information**
I couldn't find anything similar when searching, hopefully this hasn't been covered. 
This is https://www.thingiverse.com/thing:4169896. I used change at z to increase the print speed and in some cases also increase the hot end temperature. It's a speed test print. 
I've been able to successfully print other models with this version of CURA, but this is the first time I tried change at z on this version. I have NOT attempted to print this model previously. 

I'm hoping there's just something funny in my gcode.

Thanks in advance for any help!

rburema, 2020-04-01:
Does this also happen if you slice without the post-processing script? (You can see if you have a post-processing script active, because of the little hammer and wrench icon next to the slice-button area.)

mabus5, 2020-04-01:
I'll try slicing/re-printing without the post-processing when I get home from work today. My _guess_ is that it will print fine, but I will confirm.

mabus5, 2020-04-02:
Confirmed that the model does print fine when sliced with the post-processing removed. I've attached the project file and gcode for this version without scripts. 
Thanks!
[No-scripts.zip](https://github.com/Ultimaker/Cura/files/4418572/No-scripts.zip)

mahtDFR, 2020-04-03:
Good that it's solved. Closing now.

mabus5, 2020-04-03:
> Good that it's solved. Closing now.

Actually it's not solved. This isolated it to the post-processing script - it still happens when using change at z.

mahtDFR, 2020-04-06:
ah my mistake, sorry

Ghostkeeper, 2020-04-06:
From what I can see it's behaving as intended, more or less, but I think you misunderstood what is intended.

Your test here sets the print speed to 10% on layer 2. You can see it go to `F373` after layer 2 (it seems that the minimum layer time comes into play here with the initial speed, which is set to 90mm/s in the settings). This is set to "keep the change" so this transformation is applied to the entire g-code except the first 2 layers.

Then you set the print speed to 11% on layer 26. This *multiplies* with the previous change. It brings the print speed down to `F37` or about 0.6mm/s. You can probably still see it move.

Then you set the print speed to 17% on layer 51. This multiplies again, which brings the print speed down to `F6`, or about 0.1mm/s. No wonder you think that the print head stops.

At layer 76 it starts getting to `F1`, or about 0.03mm/s. This is below the default minimum speed of Marlin so it'd probably print at 0.05mm/s instead. At layer 101 and beyond it finally reaches `F0`, which would probably get ignored by the printer. Or maybe it would completely stop, I don't know.

I think you misunderstood how these changes stack up. Each of the post-processing scripts you're using will be applied separately to the g-code. If you want to go from 11% speed to 17% speed, you need to *increase* the speed to 155%. I also think that for the validity of your test you probably want to reduce the minimum layer time to 0.1s or something so that it doesn't come into effect at all.

closed 2020-04-06 by Ghostkeeper

mabus5, 2020-04-07:
Thanks for the explanation @Ghostkeeper. I did believe each change to be a percentage of the set speed and was not aware subsequent steps are a percentage of the previous speed. This makes a lot more sense now.