Ultimaker/Cura #7387: Cannot install plugins

allejp, 2020-04-01:
**Application version**
Ultimaker Cura 4.5 64bit

**Platform**
Windows 10 64bit

**Reproduction steps**
1. Click "SignIn" button and succesfully login
2. Click "Marketplace" button
3. Select one plugin
4. Click "Install" button

**Screenshot(s)**
[image]

**Actual results**
The loading icon appears over "Cancel" button but nothing happen, even after 15 minutes.

**Expected results**
The plugin is installed.

**Log file**
Logs from the last try (for this post).

```
2020-04-01 19:01:36,516 - INFO - [MainThread] Toolbox.src.Toolbox.startDownload [548]: Attempting to download & install package from https://api.ultimaker.com/cura-packages/v1/cura/v7.1.0/packages/OctoPrintPlugin/download.
2020-04-01 19:01:36,567 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [308]: Request [40c4f8736be34da6bb0acda212d4d1e2] started
2020-04-01 19:01:36,571 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-01 19:02:44,488 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/dialogs/CompatibilityDialog.qml:111:29: QML Image: Connection closed
2020-04-01 19:02:44,493 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/dialogs/CompatibilityDialog.qml:68:29: QML Image: Connection closed
2020-04-01 19:02:44,500 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:44,505 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,404 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsShowcaseTile.qml:20:5: QML Image: Connection closed
2020-04-01 19:02:47,413 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,421 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,432 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsShowcaseTile.qml:20:5: QML Image: Connection closed
2020-04-01 19:02:47,439 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,447 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,452 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,460 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsShowcaseTile.qml:20:5: QML Image: Connection closed
2020-04-01 19:02:47,465 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,470 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,476 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,481 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/pages/ToolboxAuthorPage.qml:30:9: QML Image: Connection closed
2020-04-01 19:02:47,487 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/pages/ToolboxDetailPage.qml:45:13: QML Image: Connection closed
2020-04-01 19:02:47,495 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,504 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsShowcaseTile.qml:20:5: QML Image: Connection closed
2020-04-01 19:02:47,512 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,519 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,528 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,536 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,552 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:02:47,564 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsShowcaseTile.qml:20:5: QML Image: Connection closed
2020-04-01 19:02:47,581 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:03:03,759 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:03:03,767 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:03:03,774 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:03:06,686 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:03:06,716 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
[... cut ...]
2020-04-01 19:03:06,790 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/plugins/Toolbox/resources/qml/components/ToolboxDownloadsGridTile.qml:64:9: QML Image: Connection closed
2020-04-01 19:03:16,997 - INFO - [MainThread] Toolbox.src.Toolbox.cancelDownload [566]: User cancelled the download of a package. request request[40c4f873][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/cura/v7.1.0/packages/OctoPrintPlugin/download')][timeout=None][no-data]
```

nallath, 2020-04-02:
Do you have any firewall installed? It's almost as if the download is being blocked.

allejp, 2020-04-02:
Today it worked, perhaps it was a momentary problem of the Cura servers.

closed 2020-04-02 by allejp