Ultimaker/Cura #7389: Cannot run Cura 4.5 on Catalina

thedatamachine, 2020-04-01:
**Application version**
4.5
**Platform**
iMac Retina 4k 21.5" 2017 - os 10.15.3 - graphics Radeon Pro 560 4GB
**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
Launch the app

**Screenshot(s)**
[image]
**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-04-02:
See: https://github.com/Ultimaker/Cura/issues/6700

Sadly we've still not upgraded Qt.

smartavionics, 2020-04-02:
As I mention on those linked issues, you could try installing one of my builds as they use a more recent version of Qt than the UM builds. You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there for more information.

Ghostkeeper, 2020-04-06:
Duplicate of https://github.com/Ultimaker/Cura/issues/6643.

Ghostkeeper, 2020-04-06:
Hold on, no, sorry I didn't read it properly. This is a signing issue?

Ghostkeeper, 2020-05-25:
An update I posted in a duplicate of this issue:

> The problem is that since Catalina we'd need to sign Cura twice, with both an older signing system and with a newer signing system. Apple isn't very friendly towards its developers. There are ways around it though to still run it (and then you will still run into our Qt version not properly supporting Catalina either, but it runs for the most part).

> We're still working on this, but we have been running into account management issues that even Apple Support was baffled about. I quote: "that should never happen" :smile:

fixmeplease, 2020-07-23:
Problem as above with MACOS Catalina still not allowing latest ultimaker Cura software to load still not resolved 3 months later?

Ghostkeeper, 2020-07-24:
Yup, it's still not working correctly. We've made some progress this week though since we're starting the process of getting a new certificate with the updated signature. We need a new certificate because Cura uses several features that Catalina guards against, such as DYLD environment variables (to draw system-themed dialogues) and dynamically loaded binaries (in plug-ins and for CuraEngine).