Ultimaker/Cura #7393: [master] crash on startup

smartavionics, 2020-04-02:
**Application version**
master branches as of 1645 02/04/20

**Platform**
Linux

**Printer**
Creality something...

**Reproduction steps**
Start cura

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Crashed

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**

[stderr.log](https://github.com/Ultimaker/Cura/files/4422118/stderr.log)

[cura.log](https://github.com/Ultimaker/Cura/files/4422140/cura.log)

smartavionics, 2020-04-02:
I just realised it was @fieldOfView 's PrinterSettingsPlugin that caused the crash. The LinearAdvanceSettingPlugin also crashes. Aldo, do your master branches work OK? If so, I will move to using them rather than installing from the marketplace.

closed 2020-04-02 by smartavionics

fieldOfView, 2020-04-03:
I have seen this happen with broken configurations before. Eg when some of the configuration files have been upgraded to a version that is newer than what the version of Cura that run them with expects, or a definition that is installed in the resources folder (readonly for Cura) so it cannot be upgraded by the version upgrade. Sure, those plugins seem to crash, but the crash is triggered (in my experience) by a broken configuration.

See the `cura.Settings.CuraContainerRegistry.addContainer [64]: Instance container SV01_user is outdated. Its setting version is 11 but it should be 12.` errors before the crash happens, and lots of invalid metadata errors above that.

fieldOfView, 2020-04-03:
If you don't mind sharing your configuration folder, I could have a look if there is something I can do about detecting the issue and stopping more gracefully.

smartavionics, 2020-04-03:
It's true that I had forgotten to install the upgrade to 46 plugin but even after I corrected that, I still can't use the plugins that crash. I will install one of them again and get a fresh log files.

fieldOfView, 2020-04-03:
Sigh, never mind sharing your configuration folder, I happens on my master branch too. I will have a look.

smartavionics, 2020-04-03:
It seems to be caused by trying to process/access a configuration file that (you would have thought) should have been upgraded, but hasn't?

2020-04-03 16:56:14,384 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [64]: Instance container wanhao_i3plusmk2_0.4 is outdated. Its setting ver
sion is 11 but it should be 12.

smartavionics, 2020-04-03:
the only files I can find with setting version 11 are variants from a long discarded printer?

fieldOfView, 2020-04-03:
For me it was caused by running an older branch on a previously upgraded configuration. Once I made sure I was on the current master branch, my configuration got upgraded properly again. 

Can you make sure your Cura branch is current and try again? `setting_version` should now be 13.

fieldOfView, 2020-04-03:
Resumé: this  was caused by one or more configuration files not being able to deserialize or load. In my case, and likely in @smartavionics' case too, it was caused by one or more files having a wrong `setting_version`.

I have made my plugins more robust against such cases here:
https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/commit/75b56c6045c10404c1f9aa78fed1b8fdde2f4410
https://github.com/fieldOfView/Cura-ZOffsetPlugin/commit/e082401e89108bfe94fa99f01f3335b785f41203
https://github.com/fieldOfView/Cura-PrinterSettingsPlugin/commit/644cf4829f633a33812ac7cf6d68c4e33d5a02de

smartavionics, 2020-04-03:
I don't know how I ended up with those bad variant files but with your fixes there I shouldn't have similar problems in the future. Thanks for the quick solution.