Ultimaker/Cura #7394: [4.5.0] Stringing on Retraction immediately after Ironing

Sophist-UK, 2020-04-02:
**Application version**
4.5.0

**Platform**
Windows (though this is a Gcode issue).

**Printer**
I use a Dagoma DiscoEasy 200.

**Reproduction steps**
Not sure how to reproduce - you will need for Cura to direct the printer to do a retractioin immediately following an Iron.

**Screenshot(s)**
Sorry - no.

**Actual results**
Because Ironing maintains the molten filament under higher than normal pressure, when you retract (or move over an empty space) immediately after Ironing, more than usual amount of filament is extracted causing a string. If the move following the retraction is over the ironing, adhesion to the ironed surface may occur marring the surface.

**Expected results**
Filament flow is NOT essential to ironing, and I would suggest that for a fixed period before Ironing finishes, extrusion is stopped in order to let the filament pressure disapate. I have no idea what the correct period might be - and it might depend on whether the nozzle is going on to deposit new filament or to retract / cross air afterwards.

**Project file**
Sorry - it was a big print and I just caught the strining happening. So I can't really share it and then tell you where it happened.

**Log file**
Didn't have logging on, and its too big a print to recreate the problem.

Ghostkeeper, 2020-04-06:
Basically like coasting, but specific to ironing then. Good idea.

mahtDFR, 2020-04-07:
We discussed it this morning and think it's a good idea, but took the decision to defer this feature for now. The suggestion is appreciated but we can't afford to put in the development time for it. We are open to someone making a pull request for this.

Sophist-UK, 2020-04-07:
Sorry - but whilst I have the Python skills (probably - not looked at the Cura code yet) to do the Cura bit, I don't have the C++ skills to do the CuraEngine part of this.

TommyC81, 2021-09-11:
Just a note of support for this idea. Getting terrible stringing after ironing.