Ultimaker/Cura #7395: Option to disable Tx command from start code, etc

Mrnt, 2020-04-02:
**Application version**
4.5.0

**Platform**
MacOS 10.15.3

**Printer**
Custom FFF Printer/FlashForge Dreamer

**Reproduction steps**
1. Create custom printer profile using "Custom">"Custom FFF printer" with Marlin flavor and two extruders
2. Slice an object
3. Save slice file as gcode

**Screenshot(s)**
N/A

**Actual results**
`T0` or `T1` is always included in the generated file before the custom machine_start_gcode block.

**Expected results**
On a custom printer profile I would not expect Cura to include any code prior to machine_start_gcode. 
However, I know that this has been raised before in regards to Cura adding temperature related g-code (eg https://github.com/Ultimaker/Cura/issues/3223) and it seems that the solution was to add replacement patterns to the custom code such as {material_bed_temperature} which Cura detects, resulting in it not adding the Cura default begin code.
If Cura must automatically try to add its own default begin code can there be a way to disable the tool select code by using a replacement pattern such as {initial_extruder_nr} in the printer's custom machine_start_gcode?

**Project file**
```
;FLAVOR:Marlin
;TIME:46
;Filament used: 0.0450115m, 0.0181898m
;Layer height: 0.1
;MINX:-11.9
;MINY:-46.1
;MINZ:0.3
;MAXX:11.9
;MAXY:10.3
;MAXZ:1.3
;Generated with Cura_SteamEngine 4.4.1
T0
M140 S80
M105
M190 S80
M104 S230
M104 T1 S175
M105
M109 S230
M105
M109 T1 S175
M82 ;absolute extrusion mode
;start custom gcode initialization
G90
G28
...
```

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-04-07:
Is there any reason why you wouldn't want to have it switch to the starting extruder at the start of the print, aside from consistency with how the temperature commands are handled?

Mrnt, 2020-04-07:
It actually generates an error response from the printer unfortunately.

Ghostkeeper, 2020-04-09:
Huh? Why would your firmware crash on getting a T0 command? It's a perfectly fine g-code :confounded: 

To workaround this bug you could introduce a search-and-replace post-processing script.

Mrnt, 2020-04-16:
Sorry for the delay in responding - I don't always see github notifications and the current work from home situation is less than ideal.

> Huh? Why would your firmware crash on getting a T0 command? It's a perfectly fine g-code 😖

Hahah because it's FlashForge and they made their own flavor that barfs on lots of g-code, though it works fine with all the standard move, heat etc type commands. I really don't want to run a post-processing script and want to enable all those people that have FlashForge printers using closed source firmware (ie not the FF Creator folks who are already fine) to be able to use Cura .

I would argue that the "T" command is standard to cnc/milling but historically not to 3D printing and therefore we should be able to disable it somehow, eg by using a replacement pattern in the custom profile if there can not be a checkbox option in the printer profile to disable it.

Mrnt, 2020-04-16:
I have created a plugin for Octoprint that supports all these FlashForge printers with weird g-code and works as a "print directly option from Cura" so it would be super if we can get Cura to generate code that is friendly to these printers...

Ghostkeeper, 2020-04-22:
Similarly, we'd also accept a Cura plug-in that does the same from Cura's side. However we don't really have the resources to support FlashForge printers since we can't test anything on the printer to see whether our code works.

Mrnt, 2020-04-22:
If were to make code changes, test them and submit a pull request would you accept it? 

If so how would you prefer it to be implemented - as a replacement pattern option (like the temps that can be overwritten with the custom printer begin g-code or as a option in the printer profile that allows the T command to be in generated code by default?

Mrnt, 2020-04-22:
I should note that its not just FlashForge branded printers - Dremel, some MonoPrice and PowerSpec printers seem to use FlashForge firmware.

Ghostkeeper, 2020-04-28:
> If were to make code changes, test them and submit a pull request would you accept it?
>
> If so how would you prefer it to be implemented - as a replacement pattern option (like the temps that can be [...]

I think there are two possible solutions that I'd accept:

* Some detection method that sees if the `T#` command is being used in the start g-code (maybe through use of `{initial_extruder_nr}`?), similar to how the temperature commands are detected. If the command is used it would need to tell CuraEngine to not put a `T#` command at the beginning. You could then prevent it from outputting a `T#` command at all by commenting out the command which does trigger the detection but doesn't get parsed by the firmware. This would not be my preferred solution but is acceptable since it's consistent with how the bed/print temperatures work. Look into StartSliceJob.py to see how this is currently implemented.
* A machine setting that determines whether to use the `T#` command or the `M108 T#` command to switch tools. This would have my preference since it requires less of a clutch on the side of the printer manufacturer or whoever writes the definition files.

Ghostkeeper, 2021-04-28:
We've discussed this again with the development team. We would prefer the solution that it detects whether to write `T0` or not by the presence of `{initial_extruder_nr}` in the start g-code. Although we should then also edit the existing start g-codes to place `T{initial_extruder_nr}` in front for every start g-code that has this setting key. So that's a version upgrade.

However since this only affects certain printers and not any Ultimaker printers, we can't give this issue any priority. We'll need to defer this. We can't expect a fix to come from Ultimaker any time soon, but we do recognise it as an issue for some people.

rakki9999111, 2021-09-20:
Quoth your own contributing guidelines: 

```
Submitting bug reports

...

Bugs related to supporting certain types of printers can usually not be solved by the Cura maintainers, since we don't have access to every 3D printer model in the world either. We have to rely on external contributors to fix this. **If it's something simple and obvious, such as a mistake in the start g-code, then we can directly fix it for you**, but e.g. issues with USB cable connectivity are impossible for us to debug.
```

It's very sad to see this issue still not resolved and you relying on external contributions. Don't have access to every printer in the world? Fine. But the answer fix has been provided for you

Ghostkeeper, 2021-09-27:
This is not a mistake in the start g-code of some printer. The whole point is that it's not in the start g-code, but Cura adds it by itself before outputting the start g-code.

> But the answer/fix has been provided for you.

The feature request has been provided for us, but the fix has not been provided yet, so far. It's pretty easy to write down what you want a program to do. Making it do that is a different matter.
I provided two options for a solution in my [post last year](https://github.com/Ultimaker/Cura/issues/7395#issuecomment-620473720). One of those two will have to be implemented by someone, if they want to adjust Cura to support printer firmware that doesn't deal well with `T0` commands if the printer is already on `T0`.

Until this is implemented, those users will have to make do with the workaround to add a Search And Replace post-processing script to remove it, or something similar like the possibility to make a plug-in as mentioned before.

Mrnt, 2021-12-04:
Remind me again why Cura needs to always add this before the start g-code? Couldn't you just omit it if the user has supplied a custom starting block (and allow the user to decide if they what to add T0 as part of their custom block)?

fieldOfView, 2021-12-04:
The last print might have ended with T1 as the active extruder. Adding the T0 is part of getting the printer in a "known" state.

Mrnt, 2021-12-04:
So maybe the default for the custom g-code starting block should just contain:
```
T0
```
And then users are free to keep it and add additional commands or just replace it?

Ghostkeeper, 2021-12-16:
It needs to start on the initial extruder, which is not always extruder 0. If you're doing a single extrusion print with extruder 1, you wouldn't want it to go to extruder 0. You'd have to use `{initial_extruder_nr}` at least, and hope that is just as correct as what CuraEngine knows now.

Another problem is that it would break the start-up sequence for existing printers in Cura that override the start g-code. We'd have to edit all existing start g-code overrides.

I provided two options for a solution in [my post last year](https://github.com/Ultimaker/Cura/issues/7395#issuecomment-620473720). The options there are better since they don't _require almost everyone to include something in their start g-code_. It only requires the deviant printers to add something, e.g. if they use a different command than T# to switch tools.

closed 2023-07-13 by github-actions[bot]