Ultimaker/Cura #7396: [4.5] Retract Continue PP Script Non-Functional

CCS86, 2020-04-02:
4.5.0 / Win 10 / UMO

I suspected the "retract continue" post processing script wasn't working and made a simple test:

[image]

Using this setup it is easy to look at retraction moves for increasing distances. The result is that they are all 2mm retraction moves, as defined in the print settings, and unaltered by the PP script.

Ghostkeeper, 2020-04-06:
Indeed, I can reproduce this issue with 3 cubes on the build plate. It doesn't seem to do anything at all any more.

Ghostkeeper, 2020-04-09:
I fixed this for 4.6, primarily here: https://github.com/Ultimaker/Cura/commit/9b997d421b25c1bfc56be7c68ba75fd8f6902653

This is one of those cases where I wonder how it could ever have worked, while I'm very sure that it did work at some point.

closed 2020-04-09 by Ghostkeeper

CCS86, 2020-04-09:
Thank you for looking into this. 

I downloaded the updated script and tested it out. The script is now modifying retracts. At first it looked incorrect, but I think that is a bug when relative extrusion is enabled. I don't usually use that setting, but for quickly testing stuff like this it is helpful. 

Here is a side by side of relative vs absolute extrusion with 0.05 retract continue:

[image]

CCS86, 2020-04-10:
Seems that the new version creates a much bigger issue:

https://github.com/Ultimaker/Cura/issues/7458