Ultimaker/Cura #7399: from cura 4.3 to cura 4.5 no meterials are found

kippy100, 2020-04-03:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-04-03:
Could you please actually fill in the bug template? We really need the information that it asks for to solve your issue.

kippy100, 2020-04-03:
done , thanks for the quick reply , appreicated

mahtDFR, 2020-04-03:
Can you describe what you're trying to achieve here?
According to your screenshot, you've loaded a gcode into Cura. You wont be able to slice this.

OPatrick, 2020-04-03:
Hello,
Same problem here on Ubuntu 19.04
No way to create (no new screen), duplicate or so (disabled buttons).
No way to duplicate profiles (disabled buttons).
Had Cura 3.3.2 from depots, that I uninstalled.
Downloaded Cura 4.5 app image which I can launch. But no material except default Ultimaker  Black TPU 95A.
Created custom local printer Tronxy Core XY 330*330*400
Thanks for your help

kippy100, 2020-04-04:
> Can you describe what you're trying to achieve here?
> According to your screenshot, you've loaded a gcode into Cura. You wont be able to slice this - it has already been sliced. Use an STL or similar as an input format.

hi there well i loaded a stl for slicing for my Anet A8 Printer but ut used to work now i get this errors

Ghostkeeper, 2020-04-07:
I'm unable to reproduce this issue so far. My steps are:

* Add a new printer, "Anet A8 PLUS".
* Click on the material selection drop-down.

It displays a bunch of materials there, all of the 1.75mm filaments.

[image]
(I can't actually take a screenshot of the drop-down because of Gnome for some reason.)

The same goes for the Anet A6, which is the one being used in the OP's screenshot.

The warnings you've copied from the log there are normal. It's finding some data from the plug-ins that it doesn't know what to do with, but it doesn't have to do anything with them so we can ignore that. It's unrelated to your issue.

Perhaps if you can share a project file it could shed some light on this problem you're having.

If you go into your machine settings (printer selection -> Manage Printers -> Machine Settings) on the tab for Extruder 1, what does it say at "Compatible material diameter"?

closed 2020-06-19 by no-response[bot]