Ultimaker/Cura #7400: Pause improvements

Sophist-UK, 2020-04-03:
**Is your feature request related to a problem? Please describe.**
The Pause button appears simply to pause printing immediately, leaving the print head over the print, hot and oozing filament.

Even for a short pause, this is not necessarily a good solution - but if (for example) you want to pause printing overnight, it is a terrible solution.

**Describe the solution you'd like**
1. Pause button should not pause immediately, but instead should continue printing for sufficient time to find a suitable point to stop and retract without spoiling the print. Ideally the print head would move away to allow access to the print.

As now, the print head would remain hot.

On resume we should consider whether we should somehow clean the head with a bed wipe (as a skirt does on print startup) - but this might depend on geometry (i.e. access for the head to get to Z=0).

At this point choice might be cancel, resume or a new option Suspend.

2. Suspend - this would be a new option which would cool off the print head (but not a heated bed). After Suspend, a Resume would do a pre-heat before doing a head clean as above.

Thoughts anyone?

mahtDFR, 2020-04-03:
Can you give some details about the setup you intend this for? What printer is this intended for? Are you using a serial connection? (UM machines don't really use those these days so it's unlikely the core dev team will put time into it).

> Pause button should not pause immediately, but instead should continue printing for sufficient time to find a suitable point to stop and retract without spoiling the print.

What would be a sufficient time? Where would be suitable to stop?

Sophist-UK, 2020-04-03:
I am using a Dagoma DiscoEasy 200 connected by USB.

In terms of sufficient time and where to stop, I would say as quickly as it can be done in a way that resuming the print would cause minimum imperfections in the print. What exactly the rules for that should be, I am not expert enough to say, but Cura clearly understands what it is printing second by second and the model geometry and is in principle in a good position to make a good judgement about where to stop.

A few obvious ideas: if the printer is going to do a retraction anyway, or skip over empty space anyway, that is an obvious place to pause. Otherwise, at the end of a current print type - wall, inner wall, infill section, or perhaps simply at the next sudden change of print direction.