Ultimaker/Cura #7401: Actual feedrate please

men8ifr, 2020-04-03:
**Is your feature request related to a problem? Please describe.**
When you slice you have no idea of the actual feedrate used 

**Describe the solution you'd like**
Would it be possible to show the actual feedrate in the model preview?

 Similar to https://blog.prusaprinters.org/calculator/ (scroll down to the max speed section) but have the feedrate shown at each layer on the model. 

 Also would it be possible to scale the legend only to print feedrate? It is very difficult to see the print feedrate from the colors, seeing the travel feedrate at the end of the scale is not very useful. 

**Describe alternatives you've considered**
Copying values into https://blog.prusaprinters.org/calculator/ (scroll down to the max speed section)

**Affected users and/or printers**
I think every-one will benefit that uses the feedrate preview, they can understand how the shape and size of the model, acceleration and jerk settings affect the speed achieves at different points in the model. Most people seem to think requesting 100mm/s gives 10mm/s everywhere (in fact that is what the preview actually shows) and this also eads to confusion why doubling th espeed does not halve print time. 

**Additional context**

rburema, 2020-04-04:
So the (maybe confusingly named) 'Speed' color-scheme isn't working for your purposes?

[image]

Ghostkeeper, 2020-04-07:
He's asking for two things here. The main feature request is to show the effect of accelerations and jerk in the preview. Currently it only shows the maximum (coasting) speed.

CuraEngine calculates the theoretical speeds precisely when it makes its time estimates. It knows them. However it immediately aggregates them into durations for the individual lines, and towards the end of the slice aggregates them to show the time estimates in the front-end. It doesn't communicate how fast each part of each line goes. There are four major points that might prevent us from implementing this:
* Performance of sending the slice data to the front end (i.e. the size of the layer data message).
* Performance of building the layer view data.
* Rendering performance.
* Performance and implementation when loading a .gcode file directly in the front-end.

The first 3 are a problem because this about triples the amount of layer view data; we'd have to store coordinates of each turn-over point where it starts or stops accelerating and what the velocity is at those points. In the rendering we can let the GPU interpolate the colours of these points; that's probably not a problem. I don't expect it to be a problem for CuraEngine's performance either, but for the size of the Protobuf message, the performance of building a 3D model from that protobuf message and for the rendering performance it's probably a 3-fold increase or so.

The last problem is even more drastic. If you're loading a .gcode file from disk, you'd want to see the same data too. I ported the time estimate class to Python once so that's not really a problem, but I expect that it would take a VERY long time to load a .gcode file with this feature in.

His second feature request is to (optionally) exclude the speed of travel moves from the speed colour scheme. That's a duplicate of this deferred issue: https://github.com/Ultimaker/Cura/issues/3598

Liger0, 2020-10-04:
If it gets implemented, it's probably better to let the user show both. It will help first for users who didn't really want to set them, and second it helps in understanding cases on a theorical level for issues like in #8321

qwazwsx, 2023-05-29:
I think this is worth implementing. I personally struggle a lot with reading the color scale and I feel that cura should have some way of viewing the exact print speed. As it stands, it is very difficult to tell the speed of the print especially when changing print settings that affect the maximum or minimum speed. See this example below with a model sliced with 3 different flow equalization ratios:

[image]

The highlighted line segments are all printed at the same speed, however, this is not immediately obvious when comparing between slices due to the color scale changes, making experimenting with settings that affect the min/max speed more difficult.

I see a few possible solutions:

1) Add graduations to the color scale
Graduations could help to obtain exact values especially when the scale's range is very wide
[image]

2) Add a simple color selector to the 'speed' tab of the color scheme menu
This color selector could show which speed corresponds to color as it is hovered over. 
Clicking directly on lines could change the 'color picker' to the color of the line to show that line's speed.

[image]

3) Add a speedometer to the layer print preview
The current print head velocity could be shown while the layer print preview animation is playing.