Ultimaker/Cura #7403: Print one at a time - tall objects

men8ifr, 2020-04-03:
**Is your feature request related to a problem? Please describe.**
I cannot print tall objects one at a time. I believe because CURA needs the item to fit under the gantry. However if you print objects from the front to the back on e.g. on a CR10 you can print multiple objects as tall as you want because once it finishes the first item the plate moves forward and the object never passes under the gantry. 

**Describe the solution you'd like**
Invoke a special mode if some-one selects print one at a time but the object is too tall to fit under the gantry. Then allow the object to be printed but the print order must be made front to back of build plate on XYZ printers, only one object can be printed before build plate moves forward to next object (they cannot be side by side) this mode should also remove the rear part of the print head envelope - this is not needed anymore and more objects can be fitted on the build plate. 

**Describe alternatives you've considered**
I do this manually, I set my gantry height = 400mm and make sure the print order starts at the front of the build plate and moves progressively backwards. This works vary nicely. 

**Affected users and/or printers**
Any-one who wants to print one object at a time, not I think the print one at a time option has many benefits and should be promoted as a good way to improve build quality and reduce the risk of failures, also allows you to check the 1st object is good before moving onto the next etc. 

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ghostkeeper, 2020-04-07:
Cura doesn't model out the exact shape of the gantry. There are many gantry shapes possible. Some printers have cross beams in X and Y direction. Some printers have just a single cross beam. Some printers have an H bridge. Some printers have three diagonal beams. Some printers have just a single arm holding the print head from one side.

To support ordering optimally with each of them eventually comes down to the problem of [inverse kinematics](https://en.wikipedia.org/wiki/Inverse_kinematics). This is both very complex and possibly very expensive to compute (at least in Python code). Some problems are easier to model, such as a halfplane that extends from the nozzle position in one ordinal direction. But we don't have the time to model that out for all types of gantries. So we've chosen to model the gantry as a full plane instead, which is the safest option.

I understand that you'd like us to spend more time on supporting your particular printer model, but it's one of almost 200 in Cura right now, and not one of Ultimaker's, to whom I'd need to justify spending that time. Also for Ultimaker's own printers though (which use a cross-beam gantry) we decided that modelling that out would be too complex to spend time on.

A similar - but different - feature request is to be able to print the last object even if it's taller than the gantry: https://github.com/Ultimaker/Cura/issues/6627 This is mostly an issue with how we'd make that clear to the user, given that the order is automatically optimised to maximise the available X/Y area.

Sophist-UK, 2020-04-14:
See #6627 for a discussion on a potential solution.

Reality4DEvolution, 2021-01-17:
SUGGESTION, for feature.  example, if objects are all 100 mm, and gantry height is 50 mm
could a feature be ADDED so that all objects on the bed get printed at 0-50 mm, and then all objects would get printed at the 51-100 mm range

Ghostkeeper, 2021-01-25:
It's not possible to make it just change the order of objects to do that, because objects that are 51-100mm tall still have a bottom side in the 0-50mm range.

What I think you're asking for is known (at least within the Cura developers) as "Towering". The original idea is that the nozzle itself is already protruding a few (~3) mm below the print head. So you could print 3mm of one part, then move to the other part and 3mm down and print that part for a number of layers, etc. This saves a lot of travelling and reduces stringing and scarring.

This was attempted at some point long ago. The implementation wasn't reliable back then, but it also suffered from lots of issues with the minimum layer time. Printing small parts that way isn't really nice because the previous layer hasn't cooled down yet. But the idea still has merit of course and I'd like to see it implemented some day.

Your feature request has a separate thread here: https://github.com/Ultimaker/Cura/issues/6642

Reality4DEvolution, 2021-01-25:
thanks for reply.  but, i need to clarify slightly (and i will go to the othe thread also).
example -   i want to print 4 identical objects, 100 mm high.   assume print head can go to 20 mm before colliding.  OR, assume print head can go to 10 mm before bed heating turns off.
so, use "print individuallly" as selection.  Print object 1, 0-10 mm.  object 2, 0-10 mm.   0bject 3, 0-10 mm.  ogject 4, 0-10 mm.  THEN, turn heat off.  then, print object 1 through 4, separately, 10.01 to 20 mm.  then repeat, 20.01 to 30 mm.
the consideration of where the print head would be (the approx 3mm) should not enter int the picture, because cura is already printing in order, objects that are under 10 mm (when option selected).  so, i just want this process to REPEAT, in increments of 10 mm or so.  and once the objects are high enoug that the heat bed turns off, it would not matter if they were printed singly, or in total group, because heat is turned off as the print head goes above 10 mm, or whatever.  I apoogize if this is confusing or not described well,  i don't know how to better describe it.  but, i'll go to the other htread and try posting.  thanks again