Ultimaker/Cura #7404: Select pause / z move locations

men8ifr, 2020-04-03:
**Is your feature request related to a problem? Please describe.**
Some-times print quality is poor in one specific location of the print (X and Y) , this can be traced to pauses when the printer is changing z or possibly very short pauses for other reasons. The pauses are not long or immediately obvious but enough to deteriorate print quality. 

**Describe the solution you'd like**
By default move the nozzle to the x,y, infill area  or at least away from outer walls when changing z layer or if there may be other pauses. 

**Describe alternatives you've considered**
You can try to change the print order inner/outer wall/infill  so the nozzle is not on an outer wall when it finishes but that causes other print quality issues. 

**Affected users and/or printers**
everyone

**Additional context**
(Add any other context or screenshots about the feature request here.)

rburema, 2020-04-04:
Have you considered changing the 'Z Seam Alignment' to 'User Specified' or 'Random'?

Ghostkeeper, 2020-04-07:
This feature is essentially the Z Seam settings. See [the Settings Guide](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/seam.md) for some examples and how to prevent it.

Cura should always try to move away into the infill of your model if there is any infill nearby, before starting a travel move that started from one of the walls.

men8ifr, 2020-04-08:
> Have you considered changing the 'Z Seam Alignment' to 'User Specified' or 'Random'?

That will only distribute the poor quality issues throughout the print. Also some users me included prefer a fixed z seam just would like the print quality to be better.

men8ifr, 2020-04-08:
> This feature is essentially the Z Seam settings. See [the Settings Guide](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/seam.md) for some examples and [...]

The article refers to seam not Z seam , I can't find a similar article for z seam if there is a possibility to move the nozzle to the infill?

Ghostkeeper, 2020-04-14:
> That will only distribute the poor quality issues throughout the print. Also some users me included prefer a fixed z seam just would like the print quality to be better.

Yeah Random would distribute it, creating small blips on your surface instead of a seam. But "User Specified" allows you to select the location where it seams.

> The article refers to seam not Z seam , I can't find a similar article for z seam if there is a possibility to move the nozzle to the infill?

Yes, the terminology used by Cura here is technically wrong. The settings apply to all seams, not just the seams created when making a Z move. Cura will move inside your model first before moving up, so there is no real Z seam even if printing the Infill Before Walls.