Ultimaker/Cura #7405: New infill pattern

men8ifr, 2020-04-03:
**Is your feature request related to a problem? Please describe.**
1- When printing mostly hollow items e.g a bin, open box etc none of the current infill patterns is very good, economical with filament or fast. 
2 - all current infill patterns can leave lines close to perpendicular to walls and close them. This can leave too much filament in this area and cause print quality issues. 

**Describe the solution you'd like**
Either add a pattern where infill lines radiate out from a central point in the object  (so at the walls the infill is ~90 degrees to the wall) this should work quite well for square or round objects .
Or (better solution but harder to do) allow the infill to be perpendicular to the outer wall. 

**Describe alternatives you've considered**
None - I have to make do with limitations of current infill patterns. 

**Affected users and/or printers**
any-one who may want to print hollow items e.g box, bin (not single walled), in addition it would also look great. 

**Additional context**
(Add any other context or screenshots about the feature request here.)

rburema, 2020-04-04:
Isn't that just the 'concentric' infill pattern?

Ghostkeeper, 2020-04-07:
I'm worried that this would create a giant blob of overextrusion in the middle though?

fieldOfView, 2020-04-07:
It could be like concentric with connecting lines that are perpendicular to the concentric walls and spaced equally per concentric section. That makes me worry about lots of travel moves though.

men8ifr, 2020-04-08:
> I'm worried that this would create a giant blob of overextrusion in the middle though?

This extrusion is only really meant for hollow parts, therefore not intended to print the center part

men8ifr, 2020-04-08:
> It could be like concentric with connecting lines that are perpendicular to the concentric walls and spaced equally per concentric section. That makes me worry about lots of travel moves though.

I think that is defeating the purpose quite a bit, I don't think it adds strength and like you say will slow down the print.

fieldOfView, 2020-04-09:
Perhaps you could draw a picture, since @Ghostkeeper and I apparently have the wrong idea from what you describe.

closed 2020-06-19 by no-response[bot]