Ultimaker/Cura #7407: Drag and drop from ZIP file to builder plate does not work anymore

369Martin369, 2020-04-03:
Since V4.5.0 no drag of a stl file from an zip file is possible....so far this worked

nallath, 2020-04-05:
Could you please fill in the entire bug template? It's there because we often need the information that it asks for.

369Martin369, 2020-04-07:
**Application version**
V4.5.0

**Platform**
X64
**Printer**
any

**Reproduction steps**
open Zip file
drag stl from zip to builderplate

**Screenshot(s)**

**Actual results**
builderplate remains empty

**Expected results**
object on builderplate

**Project file**
any stl file in a zip

**Log file**
really?

369Martin369, 2020-04-07:
workaround: move STL first to desktop. second drag desktopfile to builderplate

Ghostkeeper, 2020-04-07:
Honestly I think this depends more on the zip archiver application you're using. For me it never worked with the built-in "Archive Manger" from Ubuntu, but 7-zip on Windows allows you to do this. It depends on if they create a "proper" file drop event. It also doesn't work for other applications then, such as dropping files directly into gedit.

369Martin369, 2020-04-10:
normal Windows 10 X64, integrated zipper

closed 2020-06-19 by no-response[bot]

Ghostkeeper, 2020-06-29:
It seems that the Windows 10 zip archiver application (integrated with Explorer) doesn't produce normal drag-and-drop events. At least not as far as Qt can see.

Stelminator, 2023-10-25:
I went hunting on this one b/c I'd like the feature, but the best I came up with was https://stackoverflow.com/questions/5990019/drag-and-drop-files-from-a-zip-folder-into-my-window/5991432#5991432

We get values for `application/x-qt-windows-mime;value="Shell IDList Array"` and `application/x-qt-windows-mime;value="FileGroupDescriptorW"` formats from `drop.getDataAsString(<format>);`. The Shell IDList Array looks like it has the info needed, but I didn't manage to decode it successfully. ref: https://learn.microsoft.com/en-us/windows/win32/shell/clipboard?redirectedfrom=MSDN#cfstr_shellidlist

I was using Windows 11, but compressed folders go back to XP or earlier, so I doubt 10 vs. 11 makes any difference, other than 11 now understands more than just zip (they added libarchive): https://blogs.windows.com/windowsdeveloper/2023/05/23/bringing-the-power-of-ai-to-windows-11-unlocking-a-new-era-of-productivity-for-customers-and-developers-with-windows-copilot-and-dev-home/

Stelminator, 2023-12-06:
I made some progress on this tonight:
```qml
var bytes = new Uint8Array(drop.getDataAsArrayBuffer("application/x-qt-windows-mime;value=\"Shell IDList Array\""));
console.log("data", bytes);
```

```python
a = '''<copied integer list from console>'''
c = b''.join([int(x).to_bytes(1, 'little') for x in a.split(',')])
# So c is just the raw bytes we got from the drop item

# extract CIDA structure to get SHIDLIST of PIDLs
# https://learn.microsoft.com/en-us/windows/win32/api/shlobj_core/ns-shlobj_core-cida
d = shell.StringAsCIDA(c)

# first item is parent folder (this may be a path inside the archive), second item is list of children
# You can drop more than one thing; I'm just extracting the first right now
si = shell.SHCreateShellItem(d[0], None, d[1][0])

file_stream = si.BindToHandler(None, shell.BHID_Stream, pythoncom.IID_IStream)
x = file_stream.Read(4096)
# repeat Read calls until empty or some size limit
```

```python
>>> shell
<module 'win32com.shell.shell' from 'C:\\Python310\\lib\\site-packages\\win32comext\\shell\\shell.pyd'>
>>> pythoncom
<module 'pythoncom' (C:\Python310\lib\site-packages\pywin32_system32\pythoncom310.dll)>
```
I am not sure if running the postinstall script was necessary https://github.com/mhammond/pywin32#installing-globally

It's a very rough draft of what I had in my console, but it's a proof of concept. Throw in the proper error handling, and it's a strict upgrade from what we have right now (i.e., it already doesn't work).