Ultimaker/Cura #7409: Cura printing countdown 8x slow during Iron

Sophist-UK, 2020-04-03:
**Application version**
4.5.0

**Platform**
Windows

**Printer**
Dagoma DiscoEasy 200

**Reproduction steps**
Printing with ironing set on.

**Screenshot(s)**
No, sorry.

**Actual results**
On last layer, Cura shows e.g. 0:3:34 to go, and 8 second later it shows 0:3:33 to go etc.
I am guessing that the estimate for how long it takes to Iron is 8 times too little, so it decrements much slower.

**Expected results**
Time to go decreases (approximately) 1s per second.

Ghostkeeper, 2020-04-07:
Is the total printing time significantly more off than usual? I don't have a printer that accepts a USB cable when working from home to test this for you.

What I suspect might be happening is the effect of the motion buffer in your printer. Marlin (and I guess other firmwares too) keeps a buffer of pre-computed motion segments in a buffer. Usually it's up to like 16 commands or so. These commands are pre-parsed and accelerations/jerk is computed ahead of time. That way, if there are a bunch of tiny movement commands after one another, the processor will be able to keep up with the speed of the print head. However that also means that the printer has consumed 16 commands ahead of time and Cura has no way of knowing where exactly in the print the printer currently is. Cura will be a bit farther ahead than your printer.

If this is the issue, I'd expect that right before it starts ironing, your time estimate will be decreased faster. Towards the end, it will be decreased slower. This is because ironing lines are long, straight and slow. It's a single movement segment that takes a long time.

Sophist-UK, 2020-04-07:
Sorry - but this is the other way around than you suggest. The printer is working SLOWER than Cura expects, not faster.

I assume that the time remaining displayed is in the sliced Gcode. So my guess is that Cura is under-estimating how long ironing takes by a factor of 8.

Ghostkeeper, 2020-04-09:
That's what I said. I'd expect that right before ironing the time estimate will be running faster, but during ironing (towards the end of ironing if it's a big area) it will be running slower.

closed 2020-06-19 by no-response[bot]